sqfmi/Watchy #4: Using the BMA423 to trigger wakeup is slow

FWeinb, 2020-03-01:
I just saw that you updated the code and included support for accessing the BMA423 accelerometer. 
Seeing the `enableWakeupInterrupt` got me interested in using that to wakeup the device. I have successfully adapted the Button_Test example to include the ACC_INT_1 (GPIO 25) as seen in the [Schematic](https://github.com/sqfmi/Watchy-Hardware/blob/master/Watchy_Schematic.pdf) in the `esp_sleep_enable_ext1_wakeup` call which made it possible to react to it. 

The issue I am facing is that initialising the BMA Sensor on every boot takes approx. 1-2s So I thought I would only need to do it on the first hard reset of the esp32 as the BMA sensor should keep it's config across reboots but that isn't working. If I don't initialise the BMA the ext1 wakeup is triggered all the time. 
 
I looked into the implementation of the BMA423 sensor you provided and saw that you are uploading the config in the [begin()](https://github.com/sqfmi/Watchy/blob/master/src/bma.cpp#L46) call. 

One "solution" I came up with is to do the init of the BMA on the second core of the esp32 but that seams unnecessarily complicated.

Do you have any pointers what I could try next?

FWeinb, 2020-03-02:
I did some more research and found [BoschSensortec/BMA423-Sensor-API](https://github.com/BoschSensortec/BMA423-Sensor-API) which contains a much newer implementation. After adopting the BMA Class to this version (and fixing the timing functions, the `_dev.delay` pointer now expects a delay function that delays for microseconds instead of milliseconds). 

After that was done I added a boolean flag to the `begin()` function to indicate if a full initialisation to conditionally uploaded the config. 

If you are interested I could contribute my implementation to this repository (the API of the BMA class did slightly change though)

sqfmi, 2021-01-05:
Thanks! We just updated the library, any pull requests are welcome! We do plan to move to the latest implementation from Bosch, expecting sometime in Q1 2021

LeonMatthes, 2021-02-09:
The issue here is most likely that you didn't reset the interrupt.
If you read the interrupt flags from the BMA they get reset. It's a bit unintuitive...