sqfmi/Watchy #5: Out of stock?

elfranne, 2020-03-01:
The PayPal link say out of stock. Will it be back soon?

sqfmi, 2020-03-02:
Yes :)

cboddy, 2020-03-09:
Could you announce when there is more stock in this thread, please? Super keen to make an order.

cr0mbly, 2020-03-14:
> Could you announce when there is more stock in this thread, please? Super keen to make an order.

Same here, making a note for later :)

ymorayko, 2020-03-19:
Same here, waiting for it to become available.

dansmyth, 2020-03-26:
Ditto. Itching for this thing.
You did a great job.

biasedlogic, 2020-04-09:
Note #6 before you order. This repository didn't update code in past 2 months, it's all 'hello world' level code.

sqfmi, 2021-01-05:
Please visit https://watchy.sqfmi.com for full documentation, it's also back on sale here https://www.tindie.com/products/sqfmi/watchy/

Thanks!

closed 2021-01-05 by sqfmi