sqfmi/Watchy #6: Questions about Watchy

Izaic, 2020-03-05:
I was made aware of this watch by a article on CNX-software. In the post they mentioned this watch supports Micropython.

How does the micropython support work? 
How is this device charged? 
Does it already have the ability to sync over bluetooth to a phone for firmware updates and time sync?

robertio, 2020-03-05:
Hi,
What dimension in mm?
Is there case for it?
Can you please provide actual device pictures from different views? 
What specification it has ?  like weight, battery size? 

Best regards,
Robert

biasedlogic, 2020-04-08:
Don't bother. I just got it delivered. The paper-thin e-Ink glass is stuck with a piece of double-sided tape on top of an ESP32 unit, display edges and corners completely unsupported and free to catch on everything. There's no solution to mount the battery - not even a patch of double sided foam tape. The wrist strap shares a slot with the flex foil of the display - see how long will it live... and the hardest thing: there's no firmware to speak of beyond a 'hello world' - a 'clock' that hasn't got even as much of an UI as to allow to set time. Smartwatch as in smartphone-connected? forget it. It's dumber out of the box than a most basic LCD watch. Forget what's shown on sqfmi webpage.
I regret dropping any money on it, leave alone $55.

biasedlogic, 2020-04-09:
> Hi,
> What dimension in mm?

45.7 tall, 34mm wide, about 10mm thick without battery - there's no battery mount, included pouch battery is 4mm thick

> Is there case for it?

No. I may design one, unless i throw this 'watch' (which is equivalent to a 'esp32 development kit with display' if you name it properly) in the bin.

> Can you please provide actual device pictures from different views?
[image]
[image]
[image]
[image]

> What specification it has ? like weight, battery size?

Battery is 200mAh lipo pouch. Onboard charger charges it at 110mA. No battery management chips, like coulomb meter or such.

> Best regards,
> Robert

Greets
Marek

biasedlogic, 2020-04-09:
And here comes the bare PCB

[image]

Note: I have replaced the buzzer motor, the original had too short wires and was sitting over the components in the middle of the board, uptilting the display glass (to which it was glued btw, instead of the PCB):
[image]

biasedlogic, 2020-04-09:
https://www.thingiverse.com/thing:4272622

I designed a basic case for this hardware. It's no prettier than the 'watch' itself, but it shall protect the hardware while I try to figure out some use for it. I don't think I'm gonna wear it ;-) so the case will likely evolve towards some weather station or something of that sort.

robertio, 2020-04-11:
Dear Biasedlogic.
Many thanks for the very detailed and highly informative pictures!!
 I appreciate your work. So thank you!

 I am smiling to see you how you changed your mind from "garbage" to "a good toy" :)
 you spent a lot of time to enhance it. the case is a god start, for custom ones.

 I ordered [https://www.waveshare.com/product/1.54inch-e-paper-b.htm](url)
And https://www.waveshare.com/e-Paper-ESP32-Driver-Board.htm to have dev board for the future developments. 
I will see how is work.
I plan micropython to run on it.
I've some Esp8266 experience with micropython.

biasedlogic, 2020-04-11:
> Dear Biasedlogic.
> Many thanks for the very detailed and highly informative pictures!!
> I appreciate your work. So thank you!

Glad to be of help!

> I am smiling to see you how you changed your mind from "garbage" to "a good toy" :)

Well, no, I didn't. I still consider it overpriced for what it is, and useless at the advertised purpose. Was it sold by an European dealer, they'd get it shipped return and I'd have my money back already. Being how it is, I'm trying to make use of what I got, was I to decide again, I'd go for some kit with a 18650 cell, this tiny battery here won't let esp32 run for long. This system works as a clock (I wouldn't call it a watch) for a measurable time only because the RTC wakes it for milliseconds once per minute. If you try fetching weather from the web for example, it won't be ready in milliseconds, you will need seconds of active mode (get WiFi up, connect, get response, process it and display) and battery life will plummet.

robertio, 2020-05-20:
Hi.
This looks more matured, better priced, and have more hardware. (Battery management, buzzer, etc, )
Unfortunately it is not eink. 

https://www.banggood.com/LILYGO-TTGO-T-Watch-2020-ESP32-Main-Chip-1_54-Inch-Touch-Display-Programmable-Wearable-Environmental-Interaction-p-1671427.html

Thanks

robertio, 2020-05-20:
[TTGO T-Watch-2020 ESP32](https://www.banggood.com/LILYGO-TTGO-T-Watch-2020-ESP32-Main-Chip-1_54-Inch-Touch-Display-Programmable-Wearable-Environmental-Interaction-p-1671427.html)

Izaic, 2020-12-01:
> Hi.
> This (TTGO T-Watch-2020 ESP32) looks more matured, better priced, and have more hardware. (Battery management, buzzer, etc, )
> Unfortunately it is not eink.
> 
> https://www.banggood.com/LILYGO-TTGO-T-Watch-2020-ESP32-Main-Chip-1_54-Inch-Touch-Display-Programmable-Wearable-Environmental-Interaction-p-1671427.html
> 
> Thanks

I doo not recommend buying any lilygo watches either. I bought one and i was completely unable to open the device, which is how your supposed to be able to flash it. Look into the Pine64 pinetime, of which i can vouch for it's quality.

closed 2020-12-01 by Izaic