sqfmi/Watchy #10: Inaccurate step count

warlock31415, 2021-01-10:
The step count from the BMA423 is inaccurate. Just shaking my arm around increases the step count by a few 100s. 
I removed the step count function altogether from the 7-seg watchface example.

LeonMatthes, 2021-01-29:
I think this might be caused by an incorrect remapping of the BMA423 axes.

This is currently set up here:
https://github.com/sqfmi/Watchy/blob/b3e11b76ebdcc523f0b63b6ef8d02ffbaa23447f/src/Watchy.cpp#L584
``` cpp
struct bma423_axes_remap remap_data;
remap_data.x_axis = 1;
remap_data.x_axis_sign = 0;
remap_data.y_axis = 0;
remap_data.y_axis_sign = 0;
remap_data.z_axis  = 2;
remap_data.z_axis_sign  = 0;
```

But after checking with the [datasheet of the BMA423](https://www.bosch-sensortec.com/media/boschsensortec/downloads/application_notes_1/bst-mas-an032-00.pdf).
I think it must be this:
``` cpp
    struct bma423_axes_remap remap_data;
    remap_data.x_axis = 1;
    remap_data.x_axis_sign = 0xFF;
    remap_data.y_axis = 0;
    remap_data.y_axis_sign = 0xFF;
    remap_data.z_axis = 2;
    remap_data.z_axis_sign = 0xFF;
```

After changing the remapping, even Wrist Wear detection works excellent (using the newest BMA423 library, not the Watchy library).

I've added a PR for this change in #26

sqfmi, 2021-01-29:
Thanks @LeonMatthes! Merged . We're working on updating our library to use the latest: https://github.com/BoschSensortec/BMA423-Sensor-API