sqfmi/Watchy #14: Weather API settings

matveimug, 2021-01-18:
Hey, first of all - thankyou for the great product! Just got my Watchy and can't get away from tinkering with it.
I wanted to change my weather location, and was able to find the weather API settings in `Watchy/src/config.h`. My question is, how can I redefine these in my watchface? I have tried it everywhere, but it's still showing me weather from New York. The API code is tested and works in my browser. My code: 
```
    #define CITY_NAME "KURESSAARE" //if your city name has a space, replace with '+'
    #define COUNTRY_CODE "EE"
    #define OPENWEATHERMAP_APIKEY "[my API key]" //use your own API key :)
    #define OPENWEATHERMAP_URL "http://api.openweathermap.org/data/2.5/weather?q="
```
Sorry if it's something really obvious, I have no prior experience in Arduino or C++.

sqfmi, 2021-01-18:
Hey there! Glad to hear you're having fun with Watchy :)

The ```getWeatherData()``` in ```Watchy.cpp``` makes the API call to get and parse the weather data. It defaults to reading the on board RTC data if WiFi is not configured. Perhaps try some print statements in that call to see if the returned JSON is being parsed correctly.

matveimug, 2021-01-20:
Got it working by overwriting the `getWeatherData()` function in my watchface. I just assumed that I could just overwrite the macros in `config.h` somehow. Thanks for the reply!

closed 2021-01-20 by matveimug