sqfmi/Watchy #20: Question about battery voltage

peerdavid, 2021-01-26:
Hello,

first of all, thank you very much for this great project! I have one question regarding the measurement of the battery voltage: I got my Watchy and the voltage that is shown in the menu was at most (also after several h of charging) ~3.97V. Is it possible that the measurements made by my Watchy are shifted by about -0.25V? I compared it with some measurements I made between +BATT and GND (using a voltmeter) and the internal voltage was always shifted by -0.25V? 

Therefore, also the battery level in Watchy_7_SEG was only at level 1 or level 2 in my case...

Thank you very much,
David

sqfmi, 2021-01-26:
Yes this could be due to tolerance of the voltage divider resistors, as well as the calibration of the ADC. Check out https://randomnerdtutorials.com/esp32-adc-analog-read-arduino-ide/ on more options to read the ADC.

The best way to get accurate readings is to compare ADC readings with a multimeter reading (like what are you doing right now), then adjust the values accordingly. A fully charged battery should read around 4.2V, which is about what you are measuring now.

peerdavid, 2021-01-26:
Ok great, thank you very much for the clarification!

closed 2021-01-27 by sqfmi