sqfmi/Watchy #21: Can't get it to Work

pixelandfrets, 2021-01-26:
Love the product! I'm pretty new to coding...bought this to tinker with. I went through all the [setup steps](https://watchy.sqfmi.com/docs/getting-started/), but when I try to upload a watch face I keep getting errors. Here's what I'm getting trying to upload the 7_SEG face:

> Arduino: 1.8.13 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

> WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Traceback (most recent call last):
>   File "esptool.py", line 57, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports.py", line 29, in <module>
 >  File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
>   File "serial/tools/list_ports_posix.py", line 31, in <module>
>   File "/Library/Python/2.7/site-packages/PyInstaller/loader/pyimod03_importers.py", line 389, in load_module
 >  File "serial/tools/list_ports_osx.py", line 32, in <module>
> ValueError: dlsym(RTLD_DEFAULT, kIOMasterPortDefault): symbol not found
> Failed to execute script esptool
> Multiple libraries were found for "WiFi.h"
>  Used: /Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi
>  Not used: /private/var/folders/xq/lghvjj8s6xv33wfpm_x9dbm00000gn/T/AppTranslocation/7FF2DDFC-648C-4BFA-A503-CAC2DFC3A2E5/d/Arduino.app/Contents/Java/libraries/WiFi
exit status 255
> /private/var/folders/xq/lghvjj8s6xv33wfpm_x9dbm00000gn/T/AppTranslocation/7FF2DDFC-648C-4BFA-A503-> > > CAC2DFC3A2E5/d/Arduino.app/Contents/Java/arduino-builder returned 255
> Error compiling for board ESP32 Dev Module.

> This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sqfmi, 2021-01-27:
Are you on Big Sur by any chance? This looks like an error with your Arduino setup, check out https://github.com/espressif/arduino-esp32/issues/4408

pixelandfrets, 2021-01-27:
Thank you! After following several rabbit trails I've managed to get it to connect. 

Is there any documentation on setting up WiFi/Bluetooth?

sqfmi, 2021-01-27:
Awesome!

The Watchy library includes [WiFiManager ](https://github.com/tzapu/WiFiManager)by default, so you can start it up and use you phone to set your credentials. You can also hardcode your credentials and do other WiFi things from the ESP32 Arduino Library [examples](https://github.com/espressif/arduino-esp32/tree/master/libraries/WiFi/examples).

We suggest looking at our implementation in turning on the WiFi, getting the data you need, and turning the radios off. This will extend the battery life and also reduce the amount of heat generated by the ESP32 (i.e. avoid running WiFi scanning constantly)

pixelandfrets, 2021-01-27:
Fantastic. Thanks SO MUCH for your help!

closed 2021-01-27 by pixelandfrets