sqfmi/Watchy #22: Override other methods?

nathankellenicki, 2021-01-28:
Hi, loving Watchy. :)

I see in Watchy.h that only drawWatchFace is marked as virtual, but really I'd like to be able to override any of Watchy's methods. For example, I'd like to swap out `setTime()` for my own that uses NTP instead of manually setting the time, or change `getWeatherData()` to use a different provider.

Any suggestions on how to do achieve this without modifying the library?

sqfmi, 2021-01-28:
Hi there, thank you for your support!

We try to abstract the other details away to keep it simple for most users, so the only way right now would be to modify the library. We welcome suggestions (or even better, PRs!), but ```setTime()``` and ```getWeatherData()``` sounds like pretty common methods that users would want to override; we'll work on an update to address those :)

nathankellenicki, 2021-01-28:
Understood. It certainly came in handy still having access to the standard menu during development, but I was wondering at what point I'd need to fork off Watchy.h/cpp and integrate my own, and as eventually I want to replace the whole menu system. It's a great starting point to learn from. :)

nathankellenicki, 2021-01-28:
Perhaps it would be beneficial to mark all Watchy.h methods as virtual so that they can be overridden by the end user, if they want? Would be beneficial to "power users", and normal users need never touch them if they don't want to.

sqfmi, 2021-01-28:
That makes sense, we'll put it on our ToDo for next release :)

closed 2021-07-31 by sqfmi