sqfmi/Watchy #23: Watchy not connecting to Arduino IDE

IrishJourno, 2021-01-28:
Hi -- I can't get Watchy to accept an upload from the Arduino IDE under either MacOS Big Sur or Windows 10 on my Intel iMac. I have installed the CP2104 drivers under both environments, and tried varying the serial rate. In both, the Watchy's port appears to be recognized, as  cu.usbserial-XXXX in MacOS, and as COM3 in Windows. The console output under MacOS with verbose option os for upload is below, and the windows version gives a similar "Failed to connect to ESP32: Timed out waiting for packet header"
Any help you could give would be much appreciated!

Arduino: 1.8.13 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 115200, None"

WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
Sketch uses 1773169 bytes (90%) of program storage space. Maximum is 1966080 bytes.
Global variables use 58644 bytes (17%) of dynamic memory, leaving 269036 bytes for local variables. Maximum is 327680 bytes.
/Users/stephen/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0/esptool --chip esp32 --port /dev/cu.usbserial-00BFDDF4 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0xe000 /Users/stephen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/partitions/boot_app0.bin 0x1000 /Users/stephen/Library/Arduino15/packages/esp32/hardware/esp32/1.0.5-rc6/tools/sdk/bin/bootloader_qio_80m.bin 0x10000 /var/folders/c0/jxtk96zj7h96mxm_jl13vg8r0000gn/T/arduino_build_494456/7_SEG.ino.bin 0x8000 /var/folders/c0/jxtk96zj7h96mxm_jl13vg8r0000gn/T/arduino_build_494456/7_SEG.ino.partitions.bin 
esptool.py v3.0-dev
Serial port /dev/cu.usbserial-00BFDDF4
Connecting........_____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

SupaaSchnitzel, 2021-01-28:
I have the same issue occuring with mine

sqfmi, 2021-01-28:
Do you have another esp32 board by chance? Do those upload fine?

If not, could you also try just running ```esptool.py erase_flash``` in the terminal and see if that runs?

SupaaSchnitzel, 2021-01-28:
My other esp32 boards upload fine for me

IrishJourno, 2021-01-28:
When I run esptool.py erase flash, I get:

esptool.py v3.0
Found 4 serial ports
Serial port /dev/cu.usbserial-00BFDDF4
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.usbserial-00BFDDF4 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port /dev/cu.URT1
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.URT1 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port /dev/cu.SLAB_USBtoUART
/dev/cu.SLAB_USBtoUART failed to connect: [Errno 16] could not open port /dev/cu.SLAB_USBtoUART: [Errno 16] Resource busy: '/dev/cu.SLAB_USBtoUART'
Serial port /dev/cu.Bluetooth-Incoming-Port
Connecting........_____....._____....._____....._____....._____....._____....._____
/dev/cu.Bluetooth-Incoming-Port failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 4 available serial ports.

SupaaSchnitzel, 2021-01-28:
Here is my output:
```
 esptool.py erase_flash
esptool.py v3.0
Found 2 serial ports
Serial port COM15
Connecting........_____....._____....._____....._____....._____....._____....._____
COM15 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port COM1
Connecting........_____....._____....._____....._____....._____....._____....._____
COM1 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: Could not connect to an Espressif device on any of the 2 available serial ports.
```

IrishJourno, 2021-01-29:
Okay, I've found a work around, albeit a little awkward!

I plugged it into my Raspberry Pi 4. Turns out the RPi Arduino IDE has some nasty collisions with the ESP dev libraries that stops the example code from compiling unless I'm willing to do a lot of hand editing, BUT esptools can be installed and will run using the python -m option. The erase_flash command DID work here, and I was able then to plug it into my iMac system and upload an example sketch from the MacOS Arduino IDE. However I wasn't able to upload a second sketch until I plugged it back into my Pi and ran erase_flash again.

sqfmi, 2021-01-29:
That's great, sounds like there is something weird going on with the Arduino setup / USB on the Mac.
We flash every Watchy via USB and check before shipping, so there should be no issues with the flashing. Let us know if you figure out the fix with MacOS

seekerdave, 2021-02-02:
I have the same issue on a Windows 10 machine.  I also tried the erase_flash command.  Here's the output:

esptool.py v2.6
Serial port COM4
Connecting........_____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header

sqfmi, 2021-02-02:
@seekerdave have you installed the CP2104 drivers? Also make sure the correct COM port is selected.

seekerdave, 2021-02-02:
Yes, I downloaded the latest driver and am sure I have the right port.  The following details are from Device Manager:  
  Silicon Labs CP210x USB to UART Bridge (COM4)
  Driver Version: 10.1.10.103
  Driver Date: 1/8/2021

Ryomoo, 2021-02-03:
I have the same problem. Running Windows 10 with the CP2104 driver installed. 
`esptool.py erase_flash` gives the following Output.
```
esptool.py v2.6
Found 2 serial ports
Serial port COM5
Connecting........_____....._____....._____....._____....._____....._____....._____
COM5 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header
Serial port COM1
Connecting........_____....._____....._____....._____....._____....._____....._____
COM1 failed to connect: Failed to connect to Espressif device: Timed out waiting for packet header

A fatal error occurred: All of the 2 available serial ports could not connect to a Espressif device.

```

sqfmi, 2021-02-03:
@Ryomoo @seekerdave have you tried a few more times to see if any combination works? e.g. different USB port, different USB cable, not using a USB hub, lowering upload speed, closing all other programs that might be using the serial port, restarting the PC, using a different computer, etc. FYI the drivers are found here: https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers

Thanks!

Ryomoo, 2021-02-03:
I tried all of the above mentoined except the different computer. Will try that tomorrow. But my internet research says my board isn't starting in "flashing/uploading mode". I have no clue how to do that. Can you help me with that?

sqfmi, 2021-02-03:
Watchy has built in auto reset functionality, so it uses the DTR/RTS signals from serial to toggle GPIO 0 and EN to get into flash/upload mode.

seekerdave, 2021-02-03:
I replaced C1 with a much higher value (10uF) and eliminated the problem.  It seems that the DTR/RTS timing is preventing the ESP32 from entering the needed mode.  From a web search this problem seems especially prevalent on Windows 10 machines.  It would be nice if there was a software solution (perhaps a timing parameter in the esptool?).

Note: C1 on the schematic has a value of just 100nF.  I tried a 1uF (10X larger) and it worked intermittently.  The 10uF provides very reliable communication.

sqfmi, 2021-02-03:
Thanks @seekerdave, this is very useful information. Yes this seems to be prevalent with Windows and the C1 RC slows the rise time, which fixes the relative timing of RTS & DTR.  We'll explore software solutions for this (we flashed all the Watchy's using Linux)

cyc-x, 2021-06-13:
Hello, I am having some connection issues too. It looks like all my settings are correct and there is some connection on the port, the Serial Monitor opens, but nothing shows up when I try to enter any command. When uploading, I get a error :
 "Failed to execute script esptool
the selected serial port Failed to execute script esptool
 does not exist or your board is not connected"

The Watchy will sometimes flash during/after trying to upload, sometimes it will freeze.

Any ideas? Thanks!

derak-kilgo, 2021-06-15:
I just got mine and same issue. The device doesn't show up when connected which is a real bummer. 
I've tried different cables, and different computers too (win10, ubuntu 18/kernal 5.04 and mint/kernal 4.20)
I don't have other esp32 boards to test against.

Professor-Woody, 2022-09-14:
Apologies for zombifying this thread but it should be mentioned it works on OS-X as well (just reflashed using esptool.py with my mac after running into this on windows)

Hakky54, 2023-05-11:
I am also facing the same issue with Mac OS 13.3.1...

eunchurn, 2024-07-01:
> That's great, sounds like there is something weird going on with the Arduino setup / USB on the Mac. We flash every Watchy via USB and check before shipping, so there should be no issues with the flashing. Let us know if you figure out the fix with MacOS

I received the product today. It does not connect on MacOS. However, when I go into the menu, `/dev/tty.usbmodemXXXX` is recognized. I am unsure if it is an issue with MacOS or the hub. I have tried replacing all the hubs, but it is still not recognized. Additionally, it is not recognized on multiple Linux PCs(native USB ports) that I own. I have tested 10 different types of USB cables and 4 types of hubs. I have tested it on 2 Linux PCs. Lastly, the same issue occurs on both MacOS devices. In the case of MacOS, when I enter menu mode, `/dev/tty.usbmodemXXXX` is recognized, but when I try to flash it, I get the error `A fatal error occurred: This chip is ESP32-S3 not ESP32. Wrong --chip argument?`. Furthermore, in menu mode, the menu automatically moves, and at some point, the Watchy becomes unresponsive and then eventually returns to the clock screen after a while. Are you sure it was thoroughly tested before shipping?

robin-thoni, 2024-07-02:
Same here. Just received it 20 minutes ago.
The board will only briefly show up in `lsusb`, if the battery is unplugged while plugging the USB cable (so, when the esp is booting).
That's very frustrating, since this is already a replacement part for #239.

BTW, the PCB for this one (right) is clearly not the same I had on the previous one (left):
[image]
- The layout is different
- The peripheral chips are different
- The writings on the PCB are different

The first one I had was v2.0, since BTN3 was on GPIO35, according to the [pin map](https://watchy.sqfmi.com/docs/hardware#pin-map). Does that mean we just got old hardware laying around?

eunchurn, 2024-07-02:
> Same here. Just received it 20 minutes ago. The board will only briefly show up in `lsusb`, if the battery is unplugged while plugging the USB cable (so, when the esp is booting). That's very frustrating, since this is already a replacement part for #239.
> 
> BTW, the PCB for this one (right) is clearly not the same I had on the previous one (left): ![20240702_160359](https://private-user-images.githubusercontent.com/9640353/345063765-6991ff81-5041-44cf-b66b-5c0c1a513180.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTk5MzEwNjMsIm5iZiI6MTcxOTkzMDc2MywicGF0aCI6Ii85NjQwMzUzLzM0NTA2Mzc2NS02OTkxZmY4MS01MDQxLTQ0Y2YtYjY2Yi01YzBjMWE1MTMxODAuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDcwMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA3MDJUMTQzMjQzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9OTY3NGUwN2JmMTFkZDU5NzRiNzQwN2E5ZDdmZDE2MjdmYmIxOTY4OWNhNjQ1ZjIwMTA4Y2RkYWJlMjBkMmFjZCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.7cYQc34kZAjSa4u1Xy23xIcUFDc8oDIXGQVPRZfx-V0)
> 
> * The layout is different
> * The peripheral chips are different
> * The writings on the PCB are different
> 
> The first one I had was v2.0, since BTN3 was on GPIO35, according to the [pin map](https://watchy.sqfmi.com/docs/hardware#pin-map). Does that mean we just got old hardware laying around?

I have also applied for a refund regarding this matter. It doesn’t seem like this is an issue that can be resolved with software.

robin-thoni, 2024-07-02:
Just found this: https://github.com/Szybet/WatchySourcingHub/blob/main/Watchy%203.0%20review.md
So that's actually the v3 version. Which is mentioned nowhere... I absolutely have the same remark regarding the capacitor/resistor just next to the battery connector :/

robin-thoni, 2024-07-02:
Broken button, too:

https://github.com/sqfmi/Watchy/assets/9640353/9a9755f6-eda7-4e60-9d9c-0ee3a0a6ce03

sqfmi, 2024-07-05:
Please follow the latest Getting Started guide for instructions: https://watchy.sqfmi.com/docs/getting-started/

@robin-thoni This is an issue with arduino-esp32 v3.x.x. Please use v2.0.17 and the latest Watchy library v1.4.13

@eunchurn can you please report back after following the updated instructions and let us know if you still have any issues?

eunchurn, 2024-07-05:
@sqfmi Everything works well with the updated document!! I was panicking because it didn’t work with the previous document. Thank you for the quick feedback.

sqfmi, 2024-07-05:
Thanks for confirming @eunchurn! Closing.

closed 2024-07-05 by sqfmi