sqfmi/Watchy #25: GUILite / LVGL

niutech, 2021-01-28:
Is it possible to use [GUILite](https://github.com/idea4good/GuiLite) or [LVGL](https://github.com/lvgl/lvgl) to make more advanced GUI than Adafruit GFX library?

sqfmi, 2021-01-28:
Yes, you can definitely use those libraries as they are supported by the ESP32. You may need to find the right display driver, and also keep in mind e-ink screens don't have the same refresh rate as LCD screens, so the widget animations and callbacks will need to be adjusted as such.

niutech, 2021-01-28:
Thanks, but where can I find the e-paper driver for LVGL? [Here](https://github.com/lvgl/lv_drivers) I cannot find one.

sqfmi, 2021-01-30:
You can find the driver we use here: https://github.com/ZinggJM/GxEPD2/blob/master/src/epd/GxEPD2_154_D67.cpp

You will have to port it over as described here https://docs.lvgl.io/v7/en/html/porting/display.html

closed 2021-07-31 by sqfmi