sqfmi/Watchy #27: Time is incorrect - which sketch or file has parameters?

blescohier, 2021-01-30:
Hi, I'm enjoying Watchy very much. Probably I'm missing something really obvious, but the time on Watchy is 6 hours off, the month is correct but the day says 03 (today is the 30th). I changed the year offset in config.h to 1991 and now the year is correct, but the setHour() etc. in config.h seem to be pulling values from an array - where can I find parameters to set the time accurately?

Thanks!

sqfmi, 2021-01-30:
Hello, thank you for your support!

You don't need to change the offset in config.h to change the year, setting the datetime is done in ```Watchy::setTime()```:

https://github.com/sqfmi/Watchy/blob/231959144945f7999b27f71a33078e5b4dc0aa4f/src/Watchy.cpp#L346-L356

So in order to do this programmatically without using the UI, you would want to call this function (but first calling ```Wire.begin(SDA, SCL);```, but also only once, since you don't want to set the time to the same time every time Watchy wakes up. 

Also make sure the battery is plugged in as it powers the RTC.

We'll update the firmware shortly to make this easier :)

sqfmi, 2021-01-30:
We just pushed 1.2.3 https://github.com/sqfmi/Watchy/commit/88e683931c346f8670df5796d9e8e1a768727969

You can now pass a datetime string to init ```watchy.init("2021:01:30:17:13:25")``` in the format of ```YYYY:MM:DD:HH:MM:SS```

closed 2021-01-30 by sqfmi

blescohier, 2021-01-31:
Thanks! It’s all working great now… super fun.

Much appreciated!

-Bill