sqfmi/Watchy #32: Battery Problem?

slto, 2021-02-16:
Bought Watchy about a month ago.  i assembled the Watchy per instructions.  When charging, the red indicator light will be bright for the first few minutes and the it will dim.  But the battery indicator on the display never reaches 3 bars despite plugged into USB charging for many hours.  The watch display has constant problems refreshing and it's showing "ghost".  I double checked the display connection and lock and they seemed to be done correctly.

What else can I do to fix the problem?

slto, 2021-02-16:
(note I have not modified any settings, everything is factory default)

sqfmi, 2021-02-16:
Hello @slto, once the light goes dim then the battery should be fully charged. If the battery bar is only showing 2 bars it could mean the ADC calibration is a bit low. You can adjust that value [here](https://github.com/sqfmi/Watchy/blob/4abc4015096011b37d03586dfe0fe905778b881f/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp#L72). If you have access to a multimeter, you can measure to confirm the battery voltage is at 4.2V when the charging light goes dim, then adjust the value accordingly as well [here](https://github.com/sqfmi/Watchy/blob/4abc4015096011b37d03586dfe0fe905778b881f/src/Watchy.cpp#L521).

With regards to the ghosting, make sure you are using the latest version of GxEPD2 when compiling, that should fix the issue.

slto, 2021-02-19:
Ghost issue is fixed after connecting to Arduino IDE and uploading the watch face example.

However the battery is still bad, it would go from two bars to zero in a day, with WiFi off and watch idle. What is process to get a replacement battery?

LeonMatthes, 2021-02-19:
Your battery might be fine. The bars displayed on the default watchface tend to be pretty inaccurate...
Is your Watchy also loosing power/not updating after a day?
@sqfmi: Maybe take a look [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/adc.html#adc-calibration ) to improve the battery measurements. The ESP has built-in calibration from the factory that should give you pretty accurate voltage readings.

peerdavid, 2021-02-21:
I had the same problem - [Issue 20](https://github.com/sqfmi/Watchy/issues/20). Basically what I have done to measure the "real voltage" using a multimeter (measured between +BATT and GND) and compared it against the voltage measured by my watchy. I then found that in my case the voltage was shifted by -0.25V so I now simply add +0.25V to my measurements and accuracy is now ok and also the battery percent seems to be fine now. It also seems to be correct (to some degree) because until now I have never seen an outage and I used the watchy up to 3 days without charging :)

closed 2021-07-31 by sqfmi