sqfmi/Watchy #33: Classes inheriting from Watchy class seem to get constructed each update

Tythe, 2021-02-28:
Having a great time with Watchy so far but ran into an issue with variables never changing from their default values.

Modifying both the Pokemon watch face and the DOS watch face to have data the persist between `drawWatchFace()` calls doesn't seem to work. Adding a static variable to `WatchyDOS` and then incrementing it in `drawWatchFace` always results in the same value after the next update. From my investigation it looks like the constructor is being called every time the watch face updates. I'm wondering if this is a known limitation or a bug. I'll keep investigating in the mean time, thanks!

peerdavid, 2021-03-02:
When the esp32 is going into the deep sleep mode, all your variables are lost if they are not stored in the RTC memory. You can store variables in the RTC memory by simply writing ```RTC_DATA_ATTR uint16_t my_coutner = 0;```.

I hope this helps :)

Tythe, 2021-03-02:
It's option 3 use error. This is perfect, thanks!

closed 2021-03-02 by Tythe