sqfmi/Watchy #38: Need documentation on how to compile the whole library

marmat8951, 2021-05-30:
Hi sqfmi team,

This project needs a documentation on how to compile the whole library so we can test changes on it and not only changes on faces or new elements. Currently the only setup provided use the already compiled on with t;he Arduino IDE, but it would be great to be able to change code inside the library as well, instead of only on faces files.

Thanks for the great work you are doing here. I'm still loving my Watchy. <3

elad661, 2021-06-13:
I'd also appreciate some documentation. I can open the watchface examples with the Arduino IDE, but there's no .ino file for the library and the Arduino IDE can only open ino files, so I can't change basic things like the weather location

Storms-Engineering, 2021-06-22:
I believe you could override any constants that are defined or methods and write your own just like the drawWatchFace() method?

sterlingphoenix, 2021-07-29:
> I believe you could override any constants that are defined or methods and write your own just like the drawWatchFace() method?

@Storms-Engineering I tried that to connect to wifi (since without wifi setting the weather location won't really do much) but it didn't work. Do you know how to connect the thing to wifi? I've been looking for information about that for a, well, couple of days (:

Storms-Engineering, 2021-07-29:
So I ended up copying the entire source code and changing what I wanted and not extending any of watchys classes.  To connect wifi you could probably find that info in discord: https://discord.gg/ZXDegGV8E7

There is a ton of information on discord, you can search through messages and try and find what you need.  It is pretty active too.

Storms-Engineering, 2021-07-29:
> @Storms-Engineering I tried that to connect to wifi (since without wifi setting the weather location won't really do much) but it didn't work. Do you know how to connect the thing to wifi? I've been [...]

So I ended up copying the entire source code and changing what I wanted and not extending any of watchys classes.  To connect wifi you could probably find that info in discord: https://discord.gg/ZXDegGV8E7

There is a ton of information on discord, you can search through messages and try and find what you need.  It is pretty active too.

sterlingphoenix, 2021-08-01:
@Storms-Engineering after spending a few days on discord... it's just not a useful resource for me. It's incredibly busy, impossible to search for (since terms like "wifi" and "location" are going to pop up a _lot_), and nearly impossible to get responses to questions. 

This project needs _much_ better documentation. Changing stuff like your location should be straightforward, and if this requires rebuilding the library then that too should be documented somewhere (or at least linked to).

Storms-Engineering, 2021-08-01:
@sterlingphoenix Hmmmm.  Yea I could understand that.  I have an INO file that I setup here https://github.com/Storms-Engineering/Watchy/tree/master/src/Storms.  I will look and see if there is a place the community can contribute to documentation.  If you want to setup your own .ino file, I just created a new sketch in arduino and save it in the watchy /src folder.  Then I added all of the files in that folder to my sketch.  (I don't know if that was necessary or not?  But it worked *shrug*.)  

To change your location for weather look for the CITY_NAME constant in config.h.  There is some other stuff like country, and temp units.  Good luck!

sterlingphoenix, 2021-08-02:
I'll try that. I have already downloaded the entire github repo and did change the location (and other things) in config.h, but when I build/upload a watch face that seems to not get included. I see others with that issue.

In the meantime I managed to break the screen, so I'll probably not try this till the replacements get here...

rastagraffix, 2022-01-05:
What I did (paths are specific to my Mac):

1) Uninstall the Watchy library using the Library Manager.
2) cd into my Arduino work directory (/Users/lyndon/Documents/Arduino).
3) cd into the 'libraries' subdirectory.
4) git clone git@github.com/<user>/Watchy

The Arduino IDE should now find the Watchy library in its search path. It will automatically build the library components if they are missing or out of date.

Now you should be able to hack on the Watchy library source; it will rebuild any out of date files on the fly. Note that you won't be able to edit the library source using the Arduino IDE -- you'll have to use an external editor.  That's not a problem for me since I do most of my editing outside of the Arduino app anyway.