sqfmi/Watchy #40: RTC alarm not working

mipek, 2021-06-11:
I got my Watchy today but to my surprises the time is not updating. Upon further investigation it seems like no RTC alarm is ever received (ESP_SLEEP_WAKEUP_EXT0).
What should I do?

sqfmi, 2021-06-11:
Are all other features working and you were able to set the time?

mipek, 2021-06-11:
Yes setting the date and time is working fine (it just won't change), accelerometer works, BLE and WiFi seem to be working as well.

sqfmi, 2021-06-11:
Thanks, and the battery is connected as well? The RTC clock is powered by the battery.
If so, then the DS3231 RTC might be faulty. If so, send us a pic of the PCB via e-mail and we'll follow up.

mipek, 2021-06-11:
Thank you! yes the battery is connected. I've send a mail with a picture of the pcb.

PyroDevil, 2021-06-12:
I have the same problem. When I compile with watchy 1.2.5 I get this error:

```
WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
```

Could that be the issue?

Filipp1992, 2021-06-12:
Same problem here.

c-swa, 2021-06-12:
Currently having the same issue - when uploading the example code, all of the buttons work, the WiFi, Accelerometer, and Menus work. After setting the time, it will show on the watch face, but no update the time. 

I've tested rolling back the DS3232 RTC library to 1.2.12 to check and see if an earlier version would work, but the issue is still apparent.

sqfmi, 2021-06-12:
Could you try running the [rtcTimeTemp.ino](https://github.com/JChristensen/DS3232RTC/blob/master/examples/rtcTimeTemp/rtcTimeTemp.ino) example? We're trying to figure out if it's a firmware or faulty RTC

Cruelshoes360, 2021-06-12:
Having the same issue. Battery, screen, buttons, WiFi, BLE, accelerometer all work. Just the time does not update and the home screen does not automatically refresh. If I shake the watch, wait over a minute to verify it is not automatically updating, go to the menu and back to the home screen the pedometer increments. Like others RTC clock seems to be the issue.

riney, 2021-06-12:
Hello! Not to butt in but I'm having the same issue. Running `rtcTimeTemp` gives me the following output:

`14:43:18 12Jun2021 47.50C 117.50F
14:43:20 12Jun2021 47.50C 117.50F
14:43:21 12Jun2021 49.25C 120.65F
14:43:22 12Jun2021 49.25C 120.65F
14:43:23 12Jun2021 49.25C 120.65F
14:43:24 12Jun2021 49.25C 120.65F
14:43:25 12Jun2021 49.25C 120.65F
14:43:27 12Jun2021 49.25C 120.65F
14:43:28 12Jun2021 49.25C 120.65F
14:43:29 12Jun2021 49.25C 120.65F
14:43:30 12Jun2021 49.25C 120.65F
14:43:31 12Jun2021 51.00C 123.80F
14:43:32 12Jun2021 51.00C 123.80F
14:43:34 12Jun2021 51.00C 123.80F
`

A couple notes:

-  `rtcTimeTemp` delays 10 seconds between loops, but the time is incrementing about a second per iteration
- The initial time of "14:43" is what I set Watchy to, the last time I had the standard firmware on it.
- That temperature seems a bit off, too.

I tried rolling DS3232RTC back a few versions (to 1.2.7), but no difference.

Glad to try any other tests!

--John Riney

sqfmi, 2021-06-12:
Thanks @riney this was really helpful, looks like the clock is running slow for some reason. Not sure if it's the same issue for others. Let us look into the cause for this.

riney, 2021-06-13:
Thanks for looking into it! Let me know if there's any additional tests I can run for you. As an extra data point, if I increase the loop delay to 20s, the clock increments by 2 seconds every read. So it is legitimately running slow, not, say, incrementing every time you read it.

aliceafterall, 2021-06-13:
I've run the same and here's my result - no increment at all. My temps are also way crazy.
`15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F
15:55:10 12Jun2021 0.00C 32.00F`

c-swa, 2021-06-13:
Running the rtcTimeTemp.ino provided me the same results like @aliceafterall.

I decided to give it a second shot with the [TimeRTC.ino](https://github.com/JChristensen/DS3232RTC/blob/master/examples/TimeRTC/TimeRTC.ino) from the library, and my RTC started to provide at least proper output for time via the Serial.

`23:32:59.814 -> 15:08:35 12 6 2021
23:33:00.835 -> 15:08:36 12 6 2021
23:33:01.834 -> 15:08:37 12 6 2021
23:33:02.814 -> 15:08:38 12 6 2021
23:33:03.830 -> 15:08:39 12 6 2021
23:33:04.835 -> 15:08:40 12 6 2021
23:33:05.814 -> 15:08:41 12 6 2021
23:33:06.847 -> 15:08:42 12 6 2021
23:33:07.838 -> 15:08:43 12 6 2021
23:33:08.840 -> 15:08:44 12 6 2021
23:33:09.820 -> 15:08:45 12 6 2021`

riney, 2021-06-13:
There seems to be a difference depending on whether the battery is attached. Powered only by USB, my watch face seems to update about once a minute.

sqfmi, 2021-06-13:
@riney that is interesting as the RTC is powered by the battery. Make sure the battery is attached when testing out the time.

@c-swa the [TimeRTC.ino](https://github.com/JChristensen/DS3232RTC/blob/master/examples/TimeRTC/TimeRTC.ino) is using a ```delay(1000)``` call, so it may not be really showing the clock running. Do you have the battery attached as well?

PyroDevil, 2021-06-13:
For me TimeRTC.ino looks like this:

```
11:36:10 11Jun2021 0.00C 32.00F
11:36:10 11Jun2021 0.00C 32.00F
11:36:10 11Jun2021 0.00C 32.00F
11:36:10 11Jun2021 0.00C 32.00F
```

So my RTC doesn't seem to go.

mipek, 2021-06-13:
My rtcTimeTemp looks exactly like the one from PyroDevil and aliceafterall:
```
21:30:11 11Jun2021 0.00C 32.00F
21:30:11 11Jun2021 0.00C 32.00F
21:30:11 11Jun2021 0.00C 32.00F
21:30:11 11Jun2021 0.00C 32.00F
```

Filipp1992, 2021-06-13:
For me its the same: 

08:15:10 12Jun2021 0.00C 32.00F
08:15:10 12Jun2021 0.00C 32.00F
08:15:10 12Jun2021 0.00C 32.00F
08:15:10 12Jun2021 0.00C 32.00F

asbe, 2021-06-13:
I have the same issue. Pretty sure watch face is does not get call for update at all. (I successfully connected to wifi, but icon is still crossed over. Step counter at zero) . Time/temp output on serial is stuck at same value.. Is the culprit a faulty solder? Any way to verify this in software?

aliceafterall, 2021-06-13:
Regarding unplugging the battery, this indeed changes the result.
Battery unplugged, test sketch uploaded, results in actual incrementing time, and a wonky temp output:
```
23:31:31 13Feb2009 43.50C 110.30F
23:31:32 13Feb2009 46.75C 116.15F
23:31:33 13Feb2009 46.75C 116.15F
23:31:34 13Feb2009 46.75C 116.15F
23:31:35 13Feb2009 46.75C 116.15F
23:31:36 13Feb2009 46.75C 116.15F
23:31:37 13Feb2009 46.75C 116.15F
23:31:38 13Feb2009 46.75C 116.15F
23:31:40 13Feb2009 46.75C 116.15F
23:31:41 13Feb2009 46.75C 116.15F
23:31:42 13Feb2009 48.75C 119.75F
23:31:43 13Feb2009 48.75C 119.75F
```

sqfmi, 2021-06-13:
@aliceafterall interesting, so the clock seems to run OK if the battery is unplugged?

sqfmi, 2021-06-13:
Also came across this https://electronics.stackexchange.com/a/445257 , sounds similar to the issue

aliceafterall, 2021-06-13:
> @aliceafterall interesting, so the clock seems to run OK if the battery is unplugged? Do you know if the watch face would update if it was loaded?

Yes, I've just confirmed this. I have the default 7-seg loaded again and it indeed wakes and updates each minute.
Also, whereas initially the time was incrementing only by a second, it settled into properly updating (that is, the ten second delay resulted in 10 second increments).

aliceafterall, 2021-06-13:
The moment I plug the battery in:
```
23:35:21 13Feb2009 54.75C 130.55F
23:35:22 13Feb2009 54.75C 130.55F
> Battery plugged in here <
OSC is stopped!
23:35:23 13Feb2009 54.75C 130.55F
OSC is stopped!
23:35:23 13Feb2009 54.75C 130.55F
OSC is stopped!
23:35:23 13Feb2009 54.75C 130.55F
OSC is stopped!
23:35:23 13Feb2009 54.75C 130.55F
```

PyroDevil, 2021-06-13:
With only USB connected and not battery, the RTC runs for me as well and the display gets updated. But the RTC runs very slow. For me it takes over 7 minutes for one minute on the RTC. Clock/Oscillator issues?

electrokean, 2021-06-13:
I was just looking at the Watchy schematic and gerbers as my Watchy isn't due to arrive into Australia for another day or two.  I was wondering why it was decided to power the RTC from +BATT rather than +3V3?  The 3V3 regulator is always enabled, so I don't see any advantage.  Using the VBAT pin on the DS3231 as a single power supply does save a little battery power, but that could still have come from the regulated 3V3.

My concern is that the RTC_INT pin is pulled up via +BATT, which could exceed 3V3 and the maximum pin specification (3.6V) with a fully charged battery.  Thankfully R7 limits the current, but it isn't clear from the ESP32 docs how the GPIO pins have ESD protection diodes implemented and if this could cause problems.  In any case, I don't think this is the cause of the problems reported above, just an observation.

Based on the above info from @aliceafterall and others it appears the I2C interface to the DS3231 is working with or without the battery connected, presumably with the DS3231 getting powered via leakage through the I2C pullups.  Otherwise I2C errors should cause a return value of 0.

With the latest update from @aliceafterall it seems the main issue is that the oscillator stops when the VBAT pins changes state on the DS3231, so the OSF and !EOSC bits in the status and control registers likely need to be monitored and cleared to restart it.

The wildly inaccurate temperature readings some people are seeing worries me, but I imagine temperature conversions stop once the oscillator stops.

Other than the above, the other possibilities are soldering problems, fake DS3231 chips, or the MEMS resonator was damaged during ultrasonic cleaning?

aliceafterall, 2021-06-13:
> With the latest update from @aliceafterall it seems the main issue is that the oscillator stops when the power source changes on the DS3231, so the OSF and !EOSC bits in the status and control registers likely need to be monitored and cleared to restart it.
I've reset these bits, both with Wire and with the library watchy uses, with no change.
My code just zeros the status and control registers if it sees that the OSC stop flag was set.

supakeen, 2021-06-13:
> I was just looking at the Watchy schematic and gerbers as my Watchy isn't due to arrive into Australia for another day or two. I was wondering why it was decided to power the RTC from +BATT rather than +3V3? The 3V3 regulator is always enabled, so I don't see any advantage. Using the VBAT pin on the DS3231 as a single power supply does save a little battery power, but that could still have come from the regulated 3V3.
> 
> My concern is that the RTC_INT pin is pulled up via +BATT, which could exceed 3V3 and the maximum pin specification (3.6V) with a fully charged battery. Thankfully R7 limits the current, but it isn't clear from the ESP32 docs how the GPIO pins have ESD protection diodes implemented and if this could cause problems. In any case, I don't think this is the cause of the problems reported above, just an observation.
> 
> Based on the above info from @aliceafterall and others it appears the I2C interface to the DS3231 is working with or without the battery connected, presumably with the DS3231 getting powered via leakage through the I2C pullups. Otherwise I2C errors should cause a return value of 0.
> 
> With the latest update from @aliceafterall it seems the main issue is that the oscillator stops when the power source changes on the DS3231, so the OSF and !EOSC bits in the status and control registers likely need to be monitored and cleared to restart it.
> 
> The wildly inaccurate temperature readings some people are seeing worries me, but I imagine temperature conversions stop once the oscillator stops.
> 
> Other than the above, the other possibilities are soldering problems, fake DS3231 chips, or the MEMS resonator was damaged during ultrasonic cleaning?

I'd like to expand on this that if the chip is indeed accidentally being powered through the I2C bus then it could be that brownout detection isn't working and hence the reason for it running slowly when no battery being connected.

electrokean, 2021-06-13:
> I'd like to expand on this that if the chip is indeed accidentally being powered through the I2C bus then it could be that brownout detection isn't working and hence the reason for it running slowly when no battery being connected.
> 
> I do have the PCBs here but no clock module, someone could verify the truth table in the specsheet and any leakage over the bus.

Actually as @sqfmi correctly pointed out, when USB is connected then the BATT/VBAT signals are powered from the TP4054 charge chip.  But without any capacitors on the charge or RTC IC power pins there could be a fair bit of ripple or noise.

khant14, 2021-06-14:
Having the same issue. Here's a couple closeup pictures of my board: https://imgur.com/a/rJ8TUsO

The RTC test script output is similar to others here:

> 00:00:00 01Jan2000 0.00C 32.00F
00:00:00 01Jan2000 0.00C 32.00F
00:00:00 01Jan2000 0.00C 32.00F
// Battery unplugged
00:00:01 01Jan2000 50.25C 122.45F
00:00:02 01Jan2000 50.25C 122.45F
00:00:05 01Jan2000 50.25C 122.45F
00:00:06 01Jan2000 50.25C 122.45F

sqfmi, 2021-06-14:
As a temporary workaround, update to the latest v1.2.6 of the library, and add ```#define ESP_RTC``` at the top of the sketch in the example watchfaces e.g.

```
#include "Watchy_7_SEG.h"
#define ESP_RTC
```

It will switch to the internal RTC and update the watch face every ~60 seconds

aliceafterall, 2021-06-14:
For clarity, the above will allow the internal RTC to wake the watch....but the time is still read from the RTC chip, meaning the face still won't keep time without more customizations.

However the internal RTC drifts by 5%, so you'd want to implement an NTP server check-in anyway....

PyroDevil, 2021-06-14:
@sqfmi is this a software or a hardware issue? To me it currently looks like a HW issue, and people affected should probably send those boards to you to get a working replacement, right?

How are you going to handle this?

sqfmi, 2021-06-14:
We think this is most likely a hardware issue with a faulty RTC, we're going to setup a process so folks can send their boards in to get it repaired/replaced, will keep everyone posted.

sthornington, 2021-06-15:
I tried the ESP RTC define but now the watch seems to run extremely fast. Going to try reverting and see what happens.

Cruelshoes360, 2021-06-15:
> I tried the ESP RTC define but now the watch seems to run extremely fast. Going to try reverting and see what happens.

> I tried the ESP RTC define but now the watch seems to run extremely fast. Going to try reverting and see what happens.

The internal RTC is not very accurate... the reason for adding the additional RTC. Until the root issue is discovered (looking like bad RTC that will need to be replaced) you can look at updating the time on a schedule (maybe at the same time as temp update). Using the internal RTC is just a temporary work around.

wkumari, 2021-06-15:
Yup, the internal RTC is far from great -- I set mine 19 hours ago, and it has lost 17 minutes so far; thats ~1 minute per hour, or 1400 seconds per day. Still, it works...

sthornington, 2021-06-15:
Oh mine was gaining five minutes every minute. Turning it back to the onboard seems to work for me. It would definitely be nice to try to do periodic syncs with NTP servers.

wkumari, 2021-06-15:
Yeah, mine doesn't work with the onboard one - it never increments if using the onboard one.

JordyB16, 2021-06-15:
Same issue, received my Watchy today, got it assembled and followed all the 'get started' steps. Tried different watch faces. When uploading with Arduino i get the following error:  

WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).

Bit new to Arduino so don't yet know how to get to the or run ino files
All i know is time isn't changing on my watchy, i can set the time but then its stuck on that time

Cruelshoes360, 2021-06-15:
You can ignore the warning... That is not an issue.

As a sort of work around (not very accurate) you can add a #define ESP_RTC to the Watchy.cpp file and re-upload the watch face you want to use. This will have it use the internal RTC instead of the external one that is not working.

JordyB16, 2021-06-16:
And that's where my newbyness kicks in ;) Where can i find the watchy.ccp ? I do see the ccp files of the active watch face but how do i see the board itself? To keep this issue clean, anyone got links to some kind of tutorials?

wkumari, 2021-06-16:
Wherever you have the Arduino stuff installed there should be a directory called libraries, and in that there should be a directory called Watchy.
On my default install on my Mac it is in "Documents/Arduino/libraries/Watchy/src". In there there will be a config.h and Watchy.cpp.

JordyB16, 2021-06-18:
Thnx, got pointed in the right direction through Discord. Got it to work with the workaround but now its losing approx 2 minutes every hour, seems to run a bit to slow. Lost 14 minutes overnight. So we'll just have to wait for a solution bij sqfmi i guess

hn3000, 2021-06-20:
I have the RTC problem, too:

```
00:00:00 01Jan2000 0.00C 32.00F
00:00:00 01Jan2000 0.00C 32.00F
00:00:00 01Jan2000 0.00C 32.00F
00:00:00 01Jan2000 0.00C 32.00F
// unplug battery
00:00:05 01Jan2000 56.75C 134.15F
00:00:11 01Jan2000 57.25C 135.05F
00:00:17 01Jan2000 57.25C 135.05F
// reconnect battery
00:00:17 01Jan2000 57.25C 135.05F
00:00:17 01Jan2000 57.25C 135.05F
00:00:17 01Jan2000 57.25C 135.05F
00:00:17 01Jan2000 57.25C 135.05F
00:00:17 01Jan2000 57.25C 135.05F
// unplug battery
00:00:18 01Jan2000 57.25C 135.05F
00:00:24 01Jan2000 57.25C 135.05F
00:00:30 01Jan2000 57.25C 135.05F
00:00:36 01Jan2000 57.25C 135.05F
// reconnect battery
00:00:39 01Jan2000 57.75C 135.95F
00:00:39 01Jan2000 57.75C 135.95F
00:00:39 01Jan2000 57.75C 135.95F
00:00:39 01Jan2000 57.75C 135.95F
```

(Both temperatures are completely off, board is at ~36°C .)

lhanneus, 2021-06-20:
I have a correct message every 10 seconds with the rtcTimeTemp.ino code . But none of the watchy examples actually update time correctly, it can take 24hours to increase time of 30 minutes ...

SnoopJeDi, 2021-06-22:
After poking around in the ESP32 documentation[1], I discovered that there are multiple internal clocks available on the ESP32-PICO-D4. The one enabled by the above workaround is the default 150 kHz oscillator which is the lowest-power option, but has poor stability. There is also an 8 MHz oscillator which the docs say has better frequency stability. I've been using this mode for the last week or so and it definitely seems to be a better alternative, especially with temperature variation. It does require +5 μA more current in deep sleep, but the battery life is still great.

I've made the necessary changes in a [fork of the library](https://github.com/SnoopJeDi/Watchy/commit/713ec161bef25e5d985c1f1232d1053868171eaf) which should be usable through the Arduino IDE and tested it out with the 7_SEG example. If you're using this with custom code and want to keep track of which clock is being used, you might also want to know about the `rtc_clk_slow_freq_get()` and ``rtc_clk_slow_freq_get_hz()` helpers declared in the same header.

[1] config docs: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-esp32-rtc-clk-src, see also section 3.2 of the ESP32 technical manual: https://www.espressif.com/sites/default/files/documentation/esp32_technical_reference_manual_en.pdf for some general information about the SOC's clocks

hsanjuan, 2021-06-28:
> I've made the necessary changes in a [fork of the library](https://github.com/SnoopJeDi/Watchy/commit/713ec161bef25e5d985c1f1232d1053868171eaf) which should be usable through the Arduino IDE and tested it out with the 7_SEG example.

I've been testing this with an analog face (radio off). After 2 and a half days, sometimes it gests 3/4 minutes behind, now is 4 minutes ahead. Battery is at 70% or so, so seems quite ok. If we can correlate wake-up offset with temperature we might make it accurate after all.

SnoopJeDi, 2021-06-28:
> If we can correlate wake-up offset with temperature we might make it accurate after all.

This is one of the main things the DS3231M does. I'm not sure if people have really investigated pulling temperature data without the clock, but it's possible that maybe the software could do the same work? I'd much rather just replace the offending module for pursuing this level of accuracy.

That said, the workaround is probably not as accurate as it can be, because it's incrementing the minute counter and discarding seconds. @aliceafterall proposed on Discord that we could almost certainly do better if we used the [standard library](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/system_time.html#get-current-time) time-keeping functions instead. I haven't messed around with it yet, but if I do I'll be sure to post an update here.

hn3000, 2021-06-28:
+1 for esp32 standard library -- the current workaround has several flaws, among them the fact that button presses throw off the time keeping, we use the RTC chip to store the time without knowing whether it's just running slow or completely stopped and the way we wait does not take into account time spent drawing the watch face and whatever else is done between wake up and going back to deep sleep.On top of that, none of the available clock sources seem very accurate, but the runtime system can calibrate the slow clocks using the (way more precise) 40MHz clock -- and we're not making use of that ability.

SnoopJeDi, 2021-06-29:
> +1 for esp32 standard library -- the current workaround has several flaws, among them the fact that...wait does not take into account time spent drawing the watch face and whatever else is done between wake up and going back to deep sleep.

Agreed. In general I think Watchy could be doing more of the housekeeping parts of _all_ wakes (particularly the time keeping) in a [deep sleep wake stub](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/deep-sleep-stub.html) to avoid the menu issues. I suspect you're right about the timing as well, although I think the stdlib approach likely makes these fade away.

> On top of that, none of the available clock sources seem very accurate, but the runtime system can calibrate the slow clocks using the (way more precise) 40MHz clock -- and we're not making use of that ability.

`CLK_8MD256` seems pretty stable to me, if I gain/lose it appears to be a consistent drift which is probably because the offset isn't chosen in a particularly nuanced way (I wrote some logic to let me adjust it on-device and experimented across a few days). However, if there's an even better clock on-board, that would be cool to play with!

PyroDevil, 2021-06-30:
@sqfmi: Is there some update on how we can send you the watch back in order to fix the RTC?

robertimorrison, 2021-07-01:
I think I've finally addressed the DS3232 RTC interrupt-driven alarm wakeup for the ESP 32 -- I struggled with the issue for several days and could never get ALARM_2 to fire reliably every minute (at 00 seconds) to wake the ESP and update the display.

In watchy.cpp in _rtcConfig the original sqfmi github code does NOT enable ALARM_2  with the following (github code is MISSING this statement):

 RTC.alarm(ALARM_2);

SO the full set of statements to get the RTC wakeup interrupt to fire from _rtcConfig() is as follows:

  RTC.setAlarm(ALM2_EVERY_MINUTE,0,0,0); // Set ALARM_2 to fire every minute at 00 seconds
  RTC.alarm(ALARM_2); //Enable ALARM_2 -- ORIGINAL CODE NEVER Enabled the alarm!
  RTC.alarmInterrupt(ALARM_2, true); //enable ALARM_2 to fire the interrupt pin when it goes off
  RTC.squareWave(SQWAVE_NONE); //disable square wave output

My toolchain is the visualmicro add-in for Visual Studio which I believe is just a front-end for the Arduino IDE (I have the beta 2.0 Arduino IDE installed).   I suspect the sqfmi are using some other toolchain which for whatever strange reason does not require the RTC.alarm statement to actually set/enable the alarm.   

I also added RTC.begin() at the start of _rtcConfig, which the author of the DS3232 library (Jack Christensen) says initializes the I2C bus which allows the RTC to communicate with the ESP; wire.begin(SDA,SCL) up in the init method probably accomplishes the same thing, but adding RTC.begin does not appear to adversely affect anything   

The maxim DS3232 RTC claims good accuracy even without a quartz oscillator - 2.5ppm over 0 to 40C, which should be better than 1 sec / day.  I dont know if it actually achieves this.  

Before discovering the fix to enable the enable the RTC alarm, I was using the ESP timer to wake the ESP32 and also observed the large drift in wakeup times, resulting in inaccurate time display as reported by others.   I think the RTC is actually keeping accurate time, but the wakeup drifts quite a bit resulting in an inaccurate display.

aliceafterall, 2021-07-01:
Sadly, that doesn't fix anything for me - you probably had a functional chip, whereas mine doesn't even run the oscillator

truedat101, 2021-07-01:
@aliceafterall @PyroDevil  same ... my device appears to not keep time unless plugged in.  I don't want to talk about RMAs yet, but I would first like to see if the team is offering a solution.  I like the idea of this device.   Is there revision # or something that can help us identify the bad batch ?

hn3000, 2021-07-01:
Like for @aliceafterall , this doesn't change anything in my case.

wkumari, 2021-07-01:
Yes, I think that I’m also in the same boat.

I also filled in the Google form to buy a new screen and screen protector,
but haven’t heard anything back; I don’t actually *need* the screen - I’ve
got dust under my screen protector, and could not see an option to just buy
a few screen protectors, so…

W

SnoopJeDi, 2021-07-02:
> @aliceafterall @PyroDevil same ... my device appears to not keep time unless plugged in. I don't want to talk about RMAs yet, but I would first like to see if the team is offering a solution. I like the idea of this device. Is there revision # or something that can help us identify the bad batch ?

The simplest way to verify the problem right now is the `rtcTimeTemp.ino` example discussed throughout this thread. SQFMI mentioned something about users being able to verify that their device is faulty in a forthcoming firmware release.

As for batch identifiers, the chips are labeled with two fields (see [this image](https://sc04.alicdn.com/kf/H0e99ff402d3b441e88a9bb253cb8d386c.jpg)), whose semantics are unknown to me, but bad chips do appear to have these identifiers in common. These are the ones that have been reported on Discord by users with bad RTCs:

```
1613A3 009AC
1221A3 007AA
1608A3 790AB
1611A3 597AD
```

PyroDevil, 2021-07-02:
@SnoopJeDi: Well my RTC is one of those. `1611A3 597AD`

[21 more replies]