sqfmi/Watchy #41: New Watchy - OSX High Sierra - 10.13.6

MichaelFBA, 2021-06-11:
I just received watchy. When trying to upload one of the example sketches. It appears to start uploading but stops around 10% and throws this error.

Not sure if its just me, but do the buttons do anything? The screen is on and I can see full battery but everything is zero's out. ie. 00:00 time, 01 Jan 2000, 0 Steps etc etc. 
I have managed to see a menu screen once, but otherwise it feels like its bricked.

Any help would be appreciated

```
Arduino: 1.8.15 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

WARNING: library DS3232RTC claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).

esptool.py v3.0-dev
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32-PICO-D4 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, Embedded Flash, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 98:cd:ac:c3:b6:68
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 12260.9 kbit/s)...
Hash of data verified.
Compressed 18656 bytes to 12053...
Writing at 0x00001000... (100 %)
Wrote 18656 bytes (12053 compressed) at 0x00001000 in 0.2 seconds (effective 905.3 kbit/s)...
Hash of data verified.
Compressed 1783728 bytes to 1023832...
Writing at 0x00010000... (1 %)
Writing at 0x00014000... (3 %)
Writing at 0x00018000... (4 %)
Writing at 0x0001c000... (6 %)
Writing at 0x00020000... (7 %)

A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

sqfmi, 2021-06-11:
Does the upload always stop at 10%? Everything else looks good. Let us know if you are able to successfully upload anything via Arduino or if you keep getting the time out message. Sometimes a different USB port/computer might help.

MichaelFBA, 2021-06-12:
Yes or less. 
I tried on another laptop, this time Ubuntu. Getting different issue regarding missing library. I will post the error. Thanks for you help anyhow

jlewitt, 2021-06-18:
Other things to try:
* use a different cable
* set the baud rate/upload speed to a lower value (115200)

toymechqm, 2021-07-03:
@MichaelFBA Have you resolved this issue? It sounds like I am experiencing the same problem and have tried another cable. Haven't tried on another machine yet though.

MichaelFBA, 2021-07-05:
I havent had a chance to look yet sorry

sqfmi, 2021-07-31:
Closing for now, if issue persists please re-open.

closed 2021-07-31 by sqfmi