sqfmi/Watchy #42: Buttons do not seem to work

swimmadude66, 2021-06-11:
I just assembled my watchy and successfully flashed a couple watchfaces to it. I've seen the time update, but could not figure out how to set the time or connect to wifi. Eventually I pressed something which made a menu come up and was able to scroll down a couple entries, but then it disappeared. Since that moment, I have not been able to open the menu with any combination of button presses. To help me debug, I guess the first thing I want to know is "What button is supposed to open the menu?". Once I know what's supposed to do it, maybe I can have better luck troubleshooting...

luridfox, 2021-06-11:
I am having the same issues. Although I can't even get the correct time with a flashed watchface

swimmadude66, 2021-06-11:
welp, it magically started working after flashing v1.2.0 of the firmware, then flashing back to 1.2.5. Menu button is bottom left and it takes a second after you click. best of luck!

closed 2021-06-11 by swimmadude66

luridfox, 2021-06-11:
Oh awesome. Once I figure out how to do that all, I will try it. Sorry, this is for sure a learning project for me

luridfox, 2021-06-11:
Well that did work, now I just need to figure out more, like how to connect it wireless to anything. Anything I click anything in the menu it backs out of it