sqfmi/Watchy #43: I just be missing something.

luridfox, 2021-06-11:
I admittedly have a lot to learn about programming these. And I am fairly new to Arduino. 

I can't even figure out how to access the menu to turn on wifi or configure basic settings. I got a watchface on their but cannot get anything going with buttons.

PyroDevil, 2021-06-12:
I had the same issue.

The 'menu' button is on the lower left and the button on the upper left is 'back'. Maybe there is some bug in the firmware, but I think if you press the 'back' button first. the 'menu' button does not work for a while. Pressing the menu button multiple times afterwards for a while seems to work.

Professr, 2021-06-17:
There is a 5 second timeout after each button press, during which no buttons are responsive (this can be changed in the Watchy::handleButtonPress() function).  
The ESP32 seems to take about a second of holding a button to trigger the wake-from-sleep interrupt, and another second or so to actually wake up (I tested this by adding a vibration command to the wakeup and button-press handler routines).  
I don't see any obvious way to improve this, unless it was switched to use light sleep (0.8mA standby current) instead of deep sleep (10uA standby current) which would cut battery life _dramatically_ if I understand the docs correctly.

sqfmi, 2021-12-31:
Button issue should be fixed in latest releases.

closed 2021-12-31 by sqfmi