sqfmi/Watchy #44: Not updating/ not getting out of deep sleep

Filipp1992, 2021-06-12:
I followed the instructions and managed to make the watchy run.
So far I couldn't manage to make it update or show the right time. 
It looks like the real time clock is not working properly?
When I upload the code the following warning comes up:
_WARNING: library DS3232RTC-1.3.0 claims to run on avr architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s)._ 
But it still uploads the code. Might this be the problem? 
Then It shows the watch face but sticks to the shown time. And doesn't change at all. 
I reinstalled all the libraries, but didn't helped.

mipek, 2021-06-12:
I think this is related to #40

Filipp1992, 2021-06-12:
Thanks, yes that issues looks like the same. I missed it..

closed 2021-06-12 by Filipp1992