sqfmi/Watchy #45: Trouble Compiling 7 Segment Watch Face Example

rothnic, 2021-06-12:
Hi, I followed the documentation and ran into an issue where esp8266wifi.h doesn't exist. I believe I have updated each library, and checked all the settings, so I'm not sure what I'm missing. Is there a comprehensive list of libraries that are needed to be installed and known working versions? For example, WifiManager has alpha versions available that I didn't install, but I'm not sure if these are required or not.

```
Arduino: 1.8.13 (Mac OS X), Board: "ESP32 Dev Module, Disabled, Minimal SPIFFS (1.9MB APP with OTA/190KB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nroth/Library/Arduino15/packages -hardware /Users/nroth/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nroth/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nroth/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_build_637921 -warnings=none -build-cache /var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_cache_183859 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /Users/nroth/Documents/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG/7_SEG.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/nroth/Library/Arduino15/packages -hardware /Users/nroth/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/nroth/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/nroth/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=min_spiffs,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=none -ide-version=10813 -build-path /var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_build_637921 -warnings=none -build-cache /var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_cache_183859 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-97-gc752ad5-5.2.0.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0 -prefs=runtime.tools.esptool_py.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.esptool_py-3.0.0.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/esptool_py/3.0.0 -prefs=runtime.tools.mkspiffs.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/nroth/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -verbose /Users/nroth/Documents/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG/7_SEG.ino
Using board 'esp32' from platform in folder: /Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
Using core 'esp32' from platform in folder: /Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6
bash -c "[ ! -f \"/Users/nroth/Documents/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG\"/partitions.csv ] || cp -f \"/Users/nroth/Documents/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG\"/partitions.csv \"/var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_build_637921\"/partitions.csv"
bash -c "[ -f \"/var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_build_637921\"/partitions.csv ] || cp \"/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6\"/tools/partitions/min_spiffs.csv \"/var/folders/0v/0z62tzvn037cpx7drn6wp97m0000gp/T/arduino_build_637921\"/partitions.csv"
Detecting libraries used...
/Users/nroth/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-97-gc752ad5-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_trace -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/app_update -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/asio -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/bootloader_support -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/bt -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/coap -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/console -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/driver -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/efuse -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp-tls -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp32 -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_adc_cal -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_event -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_http_client -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_http_server -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_https_ota -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_https_server -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_ringbuf -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/esp_websocket_client -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/espcoredump -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/ethernet -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/expat -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/fatfs -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/freemodbus -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/freertos -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/heap -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/idf_test -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/jsmn -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/json -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/libsodium -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/log -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/lwip -I/Users/nroth/Library/Arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/mbedtls -I/Users/nro [... cut ...]
                 from /Users/nroth/Documents/Arduino/libraries/Watchy/examples/WatchFaces/7_SEG/7_SEG.ino:1:
/Users/nroth/Documents/Arduino/libraries/WiFiManager/WiFiManager.h:16:25: fatal error: ESP8266WiFi.h: No such file or directory
compilation terminated.
Using library Watchy at version 1.2.5 in folder: /Users/nroth/Documents/Arduino/libraries/Watchy 
Using library WiFiManager at version 0.16.0 in folder: /Users/nroth/Documents/Arduino/libraries/WiFiManager 
exit status 1
Error compiling for board ESP32 Dev Module.

```

rothnic, 2021-06-12:
Looks like it was the wifimanager version. The repository doesn't specify which version esp32 is supported in, but it must be these alpha versions. I updated to 2.0.3alpha and it seems to be compiling now. I recommend updating the documentation with working library versions required to compile.

sqfmi, 2021-06-14:
Thanks for the info! We'll update the docs to reflect the wifimanager version.

closed 2021-06-14 by sqfmi