sqfmi/Watchy #46: Issue with the display on watchy

nini1294, 2021-06-13:
Hi,

I just received my watchy and was assembling it. I saw that the display looked off and the top half is not updating and just the bottom half is updating. I tried to disconnect and reconnect the display, but that is not working.

[image]

Thanks,
Nishant

sqfmi, 2021-06-13:
Hi @nini1294 , it looks like you might have cracked the display during assembly. The original image from factory testing is still being shown on top, whereas the image update after assembly is happening in the bottom half of the display.

nini1294, 2021-06-13:
Hi, is there anything I can do about it? Can I get a replacement. I dont know when it might have cracked.

sqfmi, 2021-06-13:
Yes we offer replacement screens for $10 (free shipping for US) and includes the double sided tape. If you think the screen may have been damaged during shipping then you may want to go through UPS and claim for damages. If you would like to order a replacement you can fill out this form: https://forms.gle/hS3K4UnD1mdQbwnb8

closed 2021-06-14 by sqfmi