sqfmi/Watchy #48: Watchy won't do anything / not running.

wkumari, 2021-06-13:
Watch arrived with some time and date from initial testing.
As soon as I connected battery it went to 00:00 01 Jan 2000 empty battery, 0 steps. So far, so good.

However, the time never incremented, and the buttons don't do anything, so I plugged it in to charge for a while. After some time, nothing had changed on the display, so I figured there was some issue with waking up with low battery, so I downloaded and installed the libraries.

Building and flashing examples makes the screen change (e.g DOS gives me a DOS looking screen, 7-SEG gives me the original 7 segment display), but it doesn't seem to "run" - buttons don't do anything, time never increments, nothing seems to happen.

Clearly the display is connected properly, and the Arduino can reset the board, but it doesn't seem to "run". Any ideas?

khant14, 2021-06-14:
I just got my watchy yesterday and have the same issue you're having. Pressing the bottom left button pulls up a settings menu where i can set the time, wifi, etc. Even after I set the time, it does not increment. It does not seem to be waking up every minute.

spidie, 2021-06-14:
Similar issue here - the first time I connected it up, it worked and I could use buttoms - I then tried to put it onto my wifi network - this failed, and then when I reset, the buttons no longer worked

I pulled down the code, uploaded successfully, screen is updated with the new screen, but buttons not working and the clock doesn't clock! Only way I've been able to get it to do anything different is to connect it to serial without the battery and then I get 45oC on the 7 segment display version rather than 0oc

wkumari, 2021-06-14:
... and after sitting and pressing the buttons for a long time, I got the menu to appear once, but it then disappeared, and I cannot get it to show up again.

I had a brief look, and couldn't see if it is supposed to dump debug info to the serial port, and, if so, at what baud rate....

wkumari, 2021-06-14:
So, it appears that the Watchy library was just updated to  v1.2.6. I updated the library to this, rebuilt and flashed the watchy. Things seem **slightly** better - the menu/buttons now work, but they are not very responsive - I have to press the button multiple times before they register. Once I leave the menus, and return to the normal time display it seems to lock up (I set the time to 9:08, and it was stuck on that till 9:16. The buttons also no longer respond.

I think that there is something like a deep sleep which it doesn't wake up from.

wkumari, 2021-06-14:
Ah. I've found that if I hold down the button it will *sometimes* bring up the menu. When this works, it will generally work 4 or 5 times in a row, but will then refuse to work for a few minutes....

Whatever the case, the time still won't update - it's been stuck at 9:15 since I set it this morning.

wkumari, 2021-06-14:
Looking at the commit log, there was a commit (aabb888069a54fe299f068e4171917bd3f6d1c9c) with a description of "Add support for ESP RTC".

Adding:
`#define ESP_RTC 1`
to the end of config.h (before the `#endif`) enables this, and now the time seems to be running.

sthornington, 2021-06-15:
Following, mine has the same issues.

PyroDevil, 2021-06-15:
This is probably a duplicate of #40