sqfmi/Watchy #49: Watch running but buttons not responsive

SailingGreg, 2021-06-14:
Hi, the watch is working but have been unable to gain access to the configuration menu as the buttons are not responsive. Is there a trick to getting access to the menu?

SailingGreg, 2021-06-14:
I think is because of the alignment in the case - I can assess the menu sometimes if I press button 4 which pushes the watch assembly to the left and then button 1. The other option is to take the case apart and to press button 1 directly

pandalanax, 2021-06-14:
Before assembling i tested everything and the buttons worked.
However after putting the watchy together, i get the same as @SailingGreg .
Things i tried:
- USB plugged in/out
- remove the case
- charged about 30 minutes

SailingGreg, 2021-06-14:
@pandalanax I have found that you need to hold the button down for the order of a second for it to be recognised

pandalanax, 2021-06-14:
Can not confirm this..
However playing with it a bit longer revealed that out of case the button works 8/10 times with battery plugged in.
When i put it inside the case, the button won't work though. Furthermore, when i get it out of the case it needs some time before the buttons work again..
definitely far away from calling anything deterministic here

pandalanax, 2021-06-14:
wow, if you push the buttons from above and not from the side they work just fine..
seems like they don't have the right angle when pushed from the side

lhanneus, 2021-06-14:
Same for me, the buttons do not respond.

Brettevel, 2021-06-14:
I have found SW1, SW2 and SW3 to be somewhat responsive but SW4 to be entirely unresponsive and it feels physically harder to press. Has anyone else noticed this or is mine broken?

RealStickman, 2021-06-14:
For me it's SW1 that's having issues. Harder to press and pretty much unresponsive. Though the others also sometimes took multiple presses to register.

ZikZakFR, 2021-06-15:
I just received it.
I plugged in the battery and the screen then charged the battery.
The screen refreshed properly. The wiki indicates to test calling the menu by pressing switch 1 but without any info which button SW1 is...
I tried al of them and no reaction from the watch.
After 30 minutes plugged-in I unplugged it and tried the 4 buttons again but still no response from the watch.
Note that nothing else than the battery and the screen have been connected (no case, screen not glued to the pcb, etc.)

What can I be missing?

ZikZakFR, 2021-06-15:
Replying to self.
SW1 must be the bottom-left one, I pushed it several times (like a double-push) and eventually got access to the menu.
I'm not sure what the top-left button is supposed to do but the right hand-side buttons are navigational ones and the menu seems a bit laggy.
Overall I might be more an unresponsive system than a hardware glitch.

I'll now assemble everything. The wiki might need an update about the SW location and their effect.

RealStickman, 2021-06-15:
@ZikZakFR It's not that obvious, but SW1-4 are links that lead to the Hardware layout.  
https://watchy.sqfmi.com/docs/hardware#bill-of-materials  
There the switches are labeled.

ZikZakFR, 2021-06-15:
Thanks, of course with this picture it helps a lot.
So SW2 is what seems to be a back-to-home-screen.

Watch is now almost fully assembled (minus the glass) and actually with the plastic buttons the switches seems to register way better, compared to other reports above.

PyroDevil, 2021-06-15:
I initially had problems with the buttons as well. See #43

SailingGreg, 2021-06-15:
I've tried a bit of duck tape on the back of the buttons for the plastic case to take some of the slack out and that seems to improve the responsiveness for buttons 1 & 2 but has not helped with buttons 3 & 4

bserinese, 2021-06-15:
Button 4 is very unresponsive for me, even when out of the case. SW1 and SW2 work fine for moving up and down the menu, but it is difficult to enter with SW4. Took 20 minutes to change the time. I'm hoping this is a software issue and not hardware.

rainer38, 2021-06-16:
I also have problems with the buttons. While the PCB is outside the case they do work but not very good.
But as soon as i put the PCB in the case the buttons (mainly SW1) doesn't work anymore. No matter how i try to press it nothing happens.

I have to investigate what i can do and will try to directly bridge the button on the PCB and see if this works.

rainer38, 2021-06-16:
I tried to assemble the PCB and case again and now i have paid attention about the routing of the battery cable. Because there are a few positions where the cable overlaps some quite high components and i have the feeling it gets some pressure on the PCB when i screw it together. So this time i bend the cable a few times to get sure it lays quite free inside the case and now it seems much better with the responsiveness of the buttons. I can press all buttons and they do react as expected.

Interestingly SW1 is still the one i have to press the hardest to ge tit wo work and it doesn't get recognized all the times. SW2-4 seems to works quite good. So for the moment i can live with the current state of the buttons.

SailingGreg, 2021-06-17:
@sqfmi is there an issue with the switches that used on the board?

botmayank, 2021-06-19:
Hey I've also faced an issue with the switches, is there any indication if its a firmware or a hardware issue? I've re-assembled the buttons and the battery wire etc 2-3 times now, and am pretty confident nothing is damaged and assy is more or less spot-on. Even directly pressing the button (especially SW1) without the plastic part is flaky at best.

bensonk, 2021-06-19:
I got my Watchy a few days ago and have been having this problem too. When I read the thread here, I realized that it's at least partly a hardware problem. I took it apart, and found that the injection molded buttons had a bit of an overhanging edge on them, which seems to be where they were broken off of a sheet. Filing down the back of the button so it's smooth seemed to help a lot. 

I also noticed that the PCB had a similar finish -- that the flat part of the PCB the buttons press against had a bit of a jagged overhang. I very carefully filed that smooth also. 

My buttons now work much better, but I still wouldn't call them reliable. The way they operate now feels like firmware rather than hardware. I suspect there's something holding control of the CPU for longer than we'd like, which then makes it impossible for it to detect the button press.

botmayank, 2021-06-19:
My buttons look smooth and I was able to do a sanity check for continuity on the legs of the switches, seems OK. I wonder if I need to touch up the solder. At this point I think the handleButtonPress() method in https://github.com/sqfmi/Watchy/blob/master/src/Watchy.cpp is a bit bulky and leading to missed presses. 

In case anybody else wants to try, 1 and 2 are shorted and 3 and 4 are shorted. So when you press the microswitch, 1/2 should short to 3/4.
       ===       
1---| `````|---2
3---| ,,,,,,|---4

thoop, 2021-06-20:
I am only having trouble with the bottom left button.

I found a simple example of usage of the ext1 wakeup from deep sleep on the internet and changed it to use the watchy GPIO's. I am still seeing missed button presses on the menu button (buttom left) when the other buttons are always working 100% as expected. That at least points to the Watchy.cpp code not being the issue.

I saved this in a separate .ino file and uploaded this (and only this, no Watchy.cpp) to my watchy for testing:

```
/*
Deep Sleep with External Wake Up
=====================================
This code displays how to use deep sleep with
an external trigger as a wake up source and how
to store data in RTC memory to use it over reboots

This code is under Public Domain License.

Hardware Connections
======================
Push Button to GPIO 33 pulled down with a 10K Ohm
resistor

NOTE:
======
Only RTC IO can be used as a source for external wake
source. They are pins: 0,2,4,12-15,25-27,32-39.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

//original example
//#define BUTTON_PIN_BITMASK 0x8004 // GPIOs 2 and 15

#define MENU_BTN_MASK GPIO_SEL_26
#define BACK_BTN_MASK GPIO_SEL_25
#define UP_BTN_MASK GPIO_SEL_32
#define DOWN_BTN_MASK GPIO_SEL_4

#define BUTTON_PIN_BITMASK MENU_BTN_MASK|BACK_BTN_MASK|UP_BTN_MASK|DOWN_BTN_MASK

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case ESP_SLEEP_WAKEUP_EXT0 : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case ESP_SLEEP_WAKEUP_EXT1 : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case ESP_SLEEP_WAKEUP_TIMER : Serial.println("Wakeup caused by timer"); break;
    case ESP_SLEEP_WAKEUP_TOUCHPAD : Serial.println("Wakeup caused by touchpad"); break;
    case ESP_SLEEP_WAKEUP_ULP : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.printf("Wakeup was not caused by deep sleep: %d\n",wakeup_reason); break;
  }
}

/*
Method to print the GPIO that triggered the wakeup
*/
void print_GPIO_wake_up(){
  uint64_t GPIO_reason = esp_sleep_get_ext1_wakeup_status();
  Serial.print("GPIO that triggered the wake up: GPIO ");
  Serial.println((log(GPIO_reason))/log(2), 0);
}
  
void setup(){
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  //Print the GPIO used to wake up
  print_GPIO_wake_up();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up for an external trigger.
  There are two types for ESP32, ext0 and ext1 .
  ext0 uses RTC_IO to wakeup thus requires RTC peripherals
  to be on while ext1 uses RTC Controller so doesnt need
  peripherals to be powered on.
  Note that using internal pullups/pulldowns also requires
  RTC peripherals to be turned on.
  */
  //esp_deep_sleep_enable_ext0_wakeup(GPIO_NUM_15,1); //1 = High, 0 = Low

  //If you were to use ext1, you would use it like
  esp_sleep_enable_ext1_wakeup(BUTTON_PIN_BITMASK,ESP_EXT1_WAKEUP_ANY_HIGH);

  //Go to sleep now
  Serial.println("Going to sleep now");
  delay(1000);
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop(){
  //This is not going to be called
}
```
In the Serial output when running this example, the pins are:
GPIO 25 = top left
GPIO 32 = top right
GPIO 4 = bottom right
GPIO 26 = bottom left (the one with the issue)

aborzanovic, 2021-07-05:
Having a similar issue, my Watchy's bottom left button only works without the enclosure and when pressing as hard as I can. And even then it only works every 5th time. :(

albzurmar, 2021-07-29:
Same issue here, watch arrived today.
All buttons except SW1 seem to work relatively well. Without the case sometimes opening the menu works, however with the enclosure in place I can't open any menus at all.

GuruSR, 2021-07-30:
The buttons do work, though I think it is basically CPU lag happening.  The device goes into deep sleep between screen changes, I notice when it does, the buttons respond rather quickly, then not, not sure if they're causing the CPU to exit deep sleep.

Brettevel, 2021-07-30:
That would make sense, I do think the general unresponsiveness is a software issue as I noticed that they seem to be more responsive when using the 'basic' example. Although there is one button for me that is definitely physically harder to press and even when other buttons are working somewhat as expected it is barely working at all.

PrimalNaCl, 2021-07-31:
What I have been able to reproduce is that I get _ONE_ shot at the menu working after a reset.  So I have to upload a sketch (because it's in the enclosure and I'm not going through that bullshit process of cosmological planetary alignment of resassembly again) so that I can get it to reset and _THEN_ I can get into the config menu _ONE_ time.

It's _truly_ hateful.  Why does the compilation take so abysmally long.  I'm on a surface laptop 3 running win10?  Process load throughout is less than 20% and it takes a good 5 minutes to compile even the basic face.

Nurgak, 2021-08-05:
I have the same issue: new Watchy, perfectly assembled and working, but bottom left button _somewhat_ unresponsive.

When pressed hard (harder than I would like to press an SMD button from the side), I can eventually get into the menu. I don't have to press any of the other buttons as hard to register a press.

I really hope this is related to firmware. No time to investigate right now and it's not hugely problematic, but it would be nice to get to the bottom if it.

brianmay, 2021-08-05:
@Nurgak If you haven't already done so, have a look at #84. This could be your issue.

rroels, 2021-08-08:
I have the same issue, and I confirmed this is not a software issue. This is a very minimal firmware that does nothing except read/print the button state in a continuous loop:
https://pastebin.com/3p5SLU1V

My "menu" button only works about 10% of the time. For me, the click of the button also feels different compared to the other buttons. It's a lot more "mushy" with a much quieter click sound. It's just weird that it's the same button for everyone...

Nurgak, 2021-08-10:
> @Nurgak If you haven't already done so, have a look at #84. This could be your issue.

I confirm the behaviour mentioned in #84, which explains the non-responsiveness and it's definitely a bug.

However, when waking up the watch from the correct menu button I still need to press quite harder than the other buttons. It works reliably though, just need to keep in mind to press a bit harder.

andykitchen, 2021-08-13:
I loaded the button test code by @rroels from https://github.com/sqfmi/Watchy/issues/49#issuecomment-894801679 (Thanks!) I can confirm that the button hardware is unreliable and finicky:

(tested with an assembled aluminum case)

- The button will easily 'click' with both sound and haptic feedback without electrically closing the switch and triggering a press.
- Significant force is required to reliably trigger a press.
- Pressing must be focused precisely in the center of the button, a fingernail works ok, a whole finger tip is unreliable.
- Force direction must be precisely perpendicular to the case. If you come in at any angle at all the press is unreliable.
- At the hardware level, pressing for longer doesn't make a big difference, but it feels like it helps because the peak force during the long press is higher. A short sharp hard press seems to be about the same. But a longer press may be needed to register with the default watch face firmware sometimes.

These switches seem out of spec (clicking at low force, only closing at a much higher force), is it possible that the Watchy got a bad batch of switches?

Furthermore, this unreliable experience is made more frustrating by the default watch face firmware:

- There are bugs in activating the menu (#84) AND bracing against the opposite side of the case to push the menu button hard causes accidental presses of other buttons, triggering the bug.
- AFAICT there is no input buffering so a clean 'click-click' will often only move one menu item down instead of two. It seems that the first click triggers a menu update and the second click may be lost.
- Key auto-repeat gives a very mixed experience, because sometimes long presses are necessary to make the press register. Unfortunately if the watch happens to be responsive at that time the long press activates the auto-repeat and will shoot you past your desired setting, this is especially noticeable on the set time screen.

All this together means that in normal use, pressing a button gives you 0-2 presses in an unreliable and uncontrolled way which makes using the device feel very frustrating and clunky.

Workarounds I've tried:

- Experimenting with placement of the battery leads: this can help if one button works really badly. Pressure from the leads may cause the PCB to warp slightly changing the angle of force. Experimenting with several lead placements can give a small improvement, but it's wasn't a true fix for me. The above results were the best case with significant battery lead fiddling.
- Attaching tape to the back of the buttons: this helped with SW4 for me but made SW3 worse, again with some fiddling can give a small improvement but not a total fix either.

brianmay, 2021-08-13:
Wondering if some people have faulty switches. 

The only problem I have experienced here is the known software issue.  And possibly what looks very much like excessive flashing when the menu is first drawn or hidden.

andykitchen, 2021-08-13:
@brianmay: Yeah probably defective switches, difficult to say with certainty because I don't have another Watchy for reference. I just looked up the datasheet for the switch part ['K2-1114SA-A4SW-06'](https://datasheet.lcsc.com/lcsc/1810061013_Korean-Hroparts-Elec-K2-1114SA-A4SW-06_C136662.pdf) available from [LCSC here](https://lcsc.com/product-detail/Tactile-Switches_Korean-Hroparts-Elec-K2-1114SA-A4SW-06_C136662.html). AFAICT the spec'd pressing force is the highest (4) at 250+-30g although I may have decoded the part number incorrectly. At as high as 280g, this may actually be ballpark in-spec. A large pressing force means accidental presses are more unlikely, but interactive use is more difficult; so it is a tradeoff. I think this may also just be a poorly designed switch and/or has a sub-optimal mounting, because it can 'click' without making an electrical connection and is very sensitive to the pressing angle (you can press _very_ hard at some odd angles without closing the contacts).

Anecdotally, it seems that other people have got straight up broken switches. If the vendor has poor quality control, it's likely that other switches like mine are border-line.

The substantial question is: Is it worth replacing one's switches? and should they be replaced with a new instance of the same part or a different part? Some of the switches @rroels is buying from mouser are 160g switches, it will be interesting to see how they work out for him.

rroels, 2021-09-16:
Sorry for the delay. 

For me the issue was resolved by replacing the button. 

These seem to be the best fit:
https://www.mouser.be/ProductDetail/710-434351045816
https://www.mouser.be/ProductDetail/710-436351045816

I ordered both because I wasn't sure, but after receiving them I could not see any differences between them. They both have white actuators. Possibly they made a mistake and sent me 710-436351045816 twice. However, that doesn't matter. Other than the plastic bit being white, it's a perfect replacement.

shtrom, 2024-08-24:
I used those https://au.mouser.com/ProductDetail/710-436351045816, and they work!