sqfmi/Watchy #50: Support Alternative Weather API Requests

rothnic, 2021-06-14:
I live in a city that doesn't appear to have a unique name, which appears to be the [only way to configure weather data](https://github.com/sqfmi/Watchy/blob/master/src/config.h#L28) at the moment. I think supporting a city id and/or coordinates would provide much more flexibility. [See API docs here](https://openweathermap.org/current).

I went to look for my city name here: https://openweathermap.org/find?q=huntsville. You can see there are multiple returned. I found the correct one, then requested it by city id from the API. The API returned the data, but the city name was simply "Huntsville". I tried one of the other cities sharing the same name, and they also return a city name of "Huntsville". So, I don't think that requesting by city name should be used at all really, considering that it could have unexpected results.

Down the line, it would be nice to expose these kinds of settings through the menu, so all personalized settings could be completed without having to flash the device. Zip code could be pretty easy to input, given it has a small number of digits, so it could be configured similar to the time settings.

sqfmi, 2021-06-14:
Thanks for your feedback! We'll put it on the roadmap for additional features for Watchy.

As always, PRs are welcome 😊 Anyone who comes across this issue, happy to review and merge your PRs!

hyphonetics, 2021-06-16:
> Down the line, it would be nice to expose these kinds of settings through the menu, so all personalized settings could be completed without having to flash the device. Zip code could be pretty easy [...]

openweathermap has codes for each city, it'd be a nice option if you could enter in your city code instead of the name, that way you could be sure of the city you're querying

MoeTrain, 2021-06-22:
> openweathermap has codes for each city, it'd be a nice option if you could enter in your city code instead of the name, that way you could be sure of the city you're querying

tapecanvas posted a how-to for doing this exact thing on the Discord a few days back. [link to git](https://github.com/tapecanvas/WatchyDefaultFahrenheit)

pacohope, 2021-12-29:
I submitted a pull request (#132 ) that makes this more dynamic and uses openweathermap's City ID instead of text-based searching.

sqfmi, 2021-12-31:
Feature updated in v1.3.1

closed 2021-12-31 by sqfmi