sqfmi/Watchy #51: Support Network Time Lookup

rothnic, 2021-06-14:
I could be missing something, but I'd expect that a watch with wifi connectivity would use network time servers, or at least have the option to. I have been playing around with mine, and the watch went dead. This results in the time and date being reset, and it isn't the easiest thing to manually set via the menu.

There are a number of network time arduino examples and libraries like [ezTime](https://github.com/ropg/ezTime), but there could be others that would be better to use. This could be maybe implemented into a watch face, but I believe it should be part of watchy core, so that watch face designers don't have to worry about it.

rothnic, 2021-06-14:
I played around with this from the 7 seg watch face and was able to get it working. ezTime doesn't work because it has a bunch of namespace conflicts, but [this simple example](https://lastminuteengineers.com/esp32-ntp-server-date-time-tutorial/) does work for gathering the network time. I also noticed that the openweathermap API call also includes information that could be used to set the time/tz of the device, but not sure which would be preferred.

One thing I discovered is there isn't currently a clean way to update the time from a watch face. I had to review the [setTime](https://github.com/sqfmi/Watchy/blob/master/src/Watchy.cpp#L359)() watchy method and manually extract the portion [at the bottom](https://github.com/sqfmi/Watchy/blob/master/src/Watchy.cpp#L505), then modify it to suit the different data types. The tmElements_t format is used heavily in the arduino world it seems, while the method to get the local time via NTP [returns a struct with different representations in some cases](https://en.cppreference.com/w/c/chrono/tm).

I'm not sure of the best way to hook into the initialization or config menu so that it could be triggered more effectively, but to get this basic working proof of concept, I made the following changes:

Added this to Watchy_7_SEG.cpp:

```
const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = -21600; // this value depends on the tz (-6 * 60 * 60 for me)
const int   daylightOffset_sec = 3600; // i think should be either 3600 or 0, based on daylight savings use in your area

bool ntpConfig = false;
bool ntpSet = false;
```

```
void Watchy7SEG::setNetworkTime()
{
  if(!ntpSet && connectWiFi()){ // only set once and only try when connected to wifi
    if(!ntpConfig){ // only set the config once, but not sure that this matters (normally done in setup method)
      configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
      ntpConfig = true;
    }
    
    // get network time
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
      Serial.println("Failed to obtain time");
      return;
    }

    // map the struct tm into the tmElements_t struct
    tmElements_t tm0;
    tm0.Month = timeinfo.tm_mon + 1; //convert from months starting on 0 to 1
    tm0.Day = timeinfo.tm_mday;
    tm0.Year = timeinfo.tm_year - 70; //convert between 1900 and 1970 year offset
    tm0.Hour = timeinfo.tm_hour;
    tm0.Minute = timeinfo.tm_min;
    tm0.Second = timeinfo.tm_sec;

    time_t t = makeTime(tm0);
    RTC.set(t);
    
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");

    ntpSet = true; // for now set a flag so we don't do this every time, but need a better way of doing this
  }
}
```

```
void Watchy7SEG::drawTime(){
    setNetworkTime(); // add this just inside the drawTime method to trigger the update
...
```

If you want to debug this some, then you need to add the serial connection in 7_SEG.ino:

```
void setup(){
  Serial.begin(115200); // add this
  watchy.init();
}
```

Add the new method to `Watchy_7_SEG.h`:

```
class Watchy7SEG : public Watchy{
    public:
        Watchy7SEG();
        void setNetworkTime(); // add this here
...
```

sthornington, 2021-06-15:
I'm interested in this too, it was my first thought.  It would be nice if one could just put in a location (or does that come from the weather API somehow?) and then it could do DST automatically.

OlegGirko, 2021-06-15:
> I also noticed that the openweathermap API call also includes information that could be used to set the time/tz of the device, but not sure which would be preferred.

Actually, this one is even better. Instead of hardcoding your location in `config.h` file and setting time manually using buttons, it would be great to set up your location from the menu, so the firmware sets time automatically and updates it every time it checks the weather.

PyroDevil, 2021-06-15:
Maybe [gadgetbridge](https://www.gadgetbridge.org/) integration would be good, so that the time can be set by pairing with a phone.

rothnic, 2021-06-15:
> Actually, this one is even better. Instead of hardcoding your location in `config.h` file and setting time manually using buttons, it would be great to set up your location from the menu, so the firmware sets time automatically and updates it every time it checks the weather.

Yeah, the only issue I noticed for that approach was it returned `-18000` for timezone for my location (central time), but that didn't seem to match the `-21600` I needed it set it to (-18000-3600).

levimaaia, 2021-08-15:
Would it make sense to add this to src/Watchy.cpp under weatherData Watchy::getWeatherData()? That way it would only fetch the NTP time every hour instead of every minute to save battery and it would work with any watch face not just Watchy_7_SEG.cpp. Of course the method would be added to Watchy.h instead of Watchy_7_SEG.h.

brianmay, 2021-08-15:
@levimaaia have a look at #71, it does just this. But the way it does this is broken. See my comment, starting from https://github.com/sqfmi/Watchy/pull/71#issuecomment-896351579

Basically the `getLocalTime` function we use assumes the network is always up, but we keep taking wifi down to save power. So the end result is we don't use ntp for requests (except for the first request) but just use the esp32 system time instead. Which isn't particularly accurate. But the RTC clock isn't particularly accurate either.

levimaaia, 2021-08-15:
@brianmay oh, boy! This is more complicated. I didn't see #71 until now. I guess there is more work to be done, indeed.

brianmay, 2021-08-15:
@levimaaia Also see #70. Which is another ntp pull request, which is very similar, and has similar issues, but does things differently.

There are a number of things I don't like about either PR, but first priority I think it making sure it tracks ntp time correctly over time, and not just for the first query.

sqfmi, 2021-12-31:
Now available in latest release :)

closed 2021-12-31 by sqfmi