sqfmi/Watchy #52: 7-SEG temperature number ovelapping with temperature unit

etwasmitbaum, 2021-06-14:
Hello, I just updated to the latest release and noticed that with the 7-SEG Watchy face the temperature number and temperature unit are overlapping. Here is a picture of it:
[image]

sqfmi, 2021-06-14:
We need to adjust the cursor coordinates to shift the temperature text to the left a little

TimSmith714, 2021-06-15:
I edited line 106 in Watchy_7_SEG.cpp to 
```cpp
display.setCursor(135 - w, 150);
```
Felt really proud of myself and then thought I ought to check how it looks with a single digit temperature. Unfortunately there's now too big a gap between the digit and the degree symbol. 
So I think the position might need some logic based on whether the temperate is greater than 9
Looking a single digit temperature with the original code looks perfectly positioned
I'd be happy to take this on (at least to have a go C++ isn't my first language!)

etwasmitbaum, 2021-06-15:
> TimSmith714:
> So I think the position might need some logic based on whether the temperate is greater than 9

Yeah you are right, I tried to set some numbers for the Cursor and used this code:
```
//,display.getTextBounds(String(temperature), 100, 150, &x1, &y1, &w, &h);
    if(temperature > 10){
      display.setCursor(101, 150);
    } else {
      display.setCursor(133, 150);
    }
    display.println(temperature);
```
I removed the `- w` for the cursor because it did some weird things for me, like placing the temperature way too far left or right even after changing the cursor just a few pixel.
With the Code above single digit and double digit numbers are placed fine for me.

schodet, 2021-06-19:
The char 1 has a large xOffset:

```c
// bitmapOffset, width, height, xAdvance, xOffset, yOffset
    {     0,   1,   1,   9,    0,    0 }, // ' '
// ...
    {   446,  25,  39,  33,    4,  -39 }, // '0'
    {   568,   5,  35,  33,   24,  -37 }, // '1'
    {   590,  25,  39,  33,    4,  -39 }, // '2'
```

And `getTextBounds` does not count the offset, only “real” pixels. That’s why `w` does not seems correct.

To solve this, x1 also needs to be taken (and I added 4 the '2' xOffset to get to the original position):

```diff
diff --git a/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp b/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
index 7626217..1c5146a 100644
--- a/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
+++ b/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
@@ -102,8 +102,8 @@ void Watchy7SEG::drawWeather(){
     display.setFont(&DSEG7_Classic_Regular_39);
     int16_t  x1, y1;
     uint16_t w, h;
-    display.getTextBounds(String(temperature), 100, 150, &x1, &y1, &w, &h);
-    display.setCursor(155 - w, 150);
+    display.getTextBounds(String(temperature), 0, 0, &x1, &y1, &w, &h);^M
+    display.setCursor(159 - w - x1, 150);^M
     display.println(temperature);
     display.drawBitmap(165, 110, strcmp(TEMP_UNIT, "metric") == 0 ? celsius : fahrenheit, 26, 20, DARKMODE ? GxEPD_WHITE : GxEPD_BLACK);
     const unsigned char* weatherIcon;
```

etwasmitbaum, 2021-06-20:
@schodet Your changes work perfectly fine! The only issue I have is with negative temperatures with two digits. In this case, the "-" will overlap with the year on the Display. The problem here is that there is just not enough free space on the display to also show negative two digits numbers correctly.

schodet, 2021-06-22:
May be reduce font when the number do not fit?

schodet, 2021-06-22:
What about this?

```diff
diff --git a/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp b/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
index 7ce9207..b17ba0c 100644
--- a/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
+++ b/examples/WatchFaces/7_SEG/Watchy_7_SEG.cpp
@@ -102,8 +102,14 @@ void Watchy7SEG::drawWeather(){
     display.setFont(&DSEG7_Classic_Regular_39);
     int16_t  x1, y1;
     uint16_t w, h;
-    display.getTextBounds(String(temperature), 100, 150, &x1, &y1, &w, &h);
-    display.setCursor(155 - w, 150);
+    display.getTextBounds(String(temperature), 0, 0, &x1, &y1, &w, &h);^M
+    if(159 - w - x1 > 87){^M
+        display.setCursor(159 - w - x1, 150);^M
+    }else{^M
+        display.setFont(&DSEG7_Classic_Bold_25);^M
+        display.getTextBounds(String(temperature), 0, 0, &x1, &y1, &w, &h);^M
+        display.setCursor(159 - w - x1, 136);^M
+    }^M
     display.println(temperature);
     display.drawBitmap(165, 110, strcmp(TEMP_UNIT, "metric") == 0 ? celsius : fahrenheit, 26, 20, DARKMODE ? GxEPD_WHITE : GxEPD_BLACK);
     const unsigned char* weatherIcon;
```

etwasmitbaum, 2021-06-22:
Thats perfect! Thank you!

schodet, 2021-06-22:
Until merged, the issue should be left open :)

etwasmitbaum, 2021-06-22:
Whoops, my bad, thx for the info.

commit 2f6cad2f (2021-08-06): Use smaller font when temperature does not fit, closes #52

This is the case for negative temperatures (-10 for example).
commit 3680e996 (2021-08-06): Fix temperature position, closes #52
commit c34e8812 (2021-08-06): Merge pull request #67 from schodet/fix-temperature-position

Fix temperature position, closes #52
closed 2021-08-06 by sqfmi