sqfmi/Watchy #53: Stuck at Firmware Update

Cody-Coleman, 2021-06-14:
I had accidentally  gone into the menu and selected firmware update (to fast to click, screen was still refershing) and the watch is now stuck on "Bluetooth Started" where I assume it's waiting to get a firmware through the Bluetooth and to connect to something. Problem is there is no way to back out of this, cancel or exit short of pulling the batter. Not a big deal in the plastic case but if I had the aluminum one instead it would be a lot of work to open it up to pull the battery (which one must re-set the time as well at that point) 

I've accidentally done this 3 times now, and it would be nice to have some way to back out of this screen or cancel the request. The back button doesn't seem to register once you've gone into the firmware update. (which I can't find any documentation on how to even do a firmware update) .

SethMelton, 2021-06-15:
I've found that killing your bluetooth connection is enough to kick it out of the waiting state.

Furthermoore, 2021-06-15:
@SethMelton what connection?   It just says 'waiting for connection...'

SethMelton, 2021-06-15:
Just forget the watchy from whatever device you've connected it to. That will force the watch to error out and let you back to the menu

Cody-Coleman, 2021-06-15:
Can confirm, I can get out of this menu if I pair with my phone, then disconnect the pair. Can't seem to detect the watchy from the macbook via bluetooth but this works to get it un-stuck.

closed 2021-06-16 by Cody-Coleman

tlhintoq, 2022-07-30:
@Cody-Coleman 
I just did the same as the original poster.  
Just walking down the main menu in order as one would expect... 

1. set the time... 
2. connect to wifi...
3.  sync to NTP... 
4. All good.  Then... 
5. Oh, update firmware.  Well that makes sense.  {guessing} Probably does that over Wifi now that its on my network.
6. Go to website on my phone per instruction
7. Click menu to go to bluetooth pairing
8. Stuck waiting for connection.  

Problem - You can't forget it from your phone if it never makes the connection in the first place.

Watch display:

Bluetooth Started
Watchy BLE OTA
Waiting for 
connection...

And now I'm hosed.  Does not appear as a device to connect to from my iPhone13maxpro, nor my MSI GS76-Stealth laptop.
Can't connect to it so can't unconnect from it.
Maybe it should time out just like the Wifi configuration did. Or maybe the 'back' button should actually go 'back' at this point. Or maybe a 10second hold on the main menu button should reboot it.  Something.

So I can either open the case, pull the battery and make sure to not do that again or... When my Arduino installation is done downloading the esp32 board and I can compile etc., maybe I can force the Watchy to reset via software/USB.

PR #180 (2025-08-19): Add BLE connection timeout
add waiting for BLE connection (timeout 60 sec)
this fixes #53, #176