sqfmi/Watchy #54: Time is 2 minutes off after one day

etwasmitbaum, 2021-06-15:
Hello, I set my Watchy time manually in the settings about 20 hours ago. Now I noticed that the time is about 2 minutes too early. Here is a picture of it:
My PC clock is synced, and I set the Watchy time with my PC. 
[image]

adamm, 2021-06-19:
Personally, I haven't seen this with the RTC, but I purposely set my watch time about a minute ahead of realtime.  It's remained roughly in sync all weekend.

I think there are three things are needed to solve your issue.

1. The ESP sleep timer needs to not default to 60000000us, and instead calculate the microseconds until the next zero-second minute, less the time it takes to wake up and actually refresh the display.
2. If Wifi is set, default to an NTP server that refreshes itself once a day.
3. The user will still need to set their timezone.  The "Set Time" menu option can be used for that.

Personally, I was able to implement 1 and 2 on my https://github.com/adamm/esp8266-clockradio project.  Should be trivial to port it over to here.

I can try to port the concept over to this project this weekend and submit a PR.

etwasmitbaum, 2021-06-19:
The ESP sleep timer by default is handled with the DS3231 which is probably broken in my watch. 
If you add `#define ESP_RTC 1` in the `config.h` file it will use the RTC of the ESP32 which is very unprecise. But you are right, you can set the sleep timer to a calculated value so its exactly one minute.
Also I actually tried myself to implement an NTP server connection, but I am still struggling with battery issues while using it.

Would be great if you can implement 1. and 2. and create a PR. Probably your solution would be better than mine is.

OverThinker, 2021-06-20:
> The ESP sleep timer by default is handled with the DS3231 which is probably broken in my watch.
> If you add `#define ESP_RTC 1` in the `config.h` file it will use the RTC of the ESP32 which is very [...]

If you look in the developer discord channel, I (AaronB) posted my ntp code.  It is working well for me so far updating the clock three times a day.

etwasmitbaum, 2021-06-20:
@OverThinker I found your code on Discord and it looks great. I tried to implement it but I am getting an error `struct tm timeinfo;`.
The error is: 
```
src/Watchy_7_SEG.cpp: In member function 'void Watchy7SEG::syncNtpTime()':
src/Watchy_7_SEG.cpp:33:15: error: aggregate 'tm timeinfo' has incomplete type and cannot be defined
     struct tm timeinfo;
               ^
```
Any idea what I am doing wrong?

etwasmitbaum, 2021-06-21:
@OverThinker Your code is working great! Thanks!

closed 2021-06-21 by etwasmitbaum

schodet, 2021-06-22:
@etwasmitbaum can you open a pull request?

schodet, 2021-06-22:
Also, this is a workaround but the RTC is not supposed to have a drift like this, 2 minutes a day is an error of 1388 ppm, while the DS3231M is supposed to have an accuracy of 2 ppm. There is something wrong with the RTC.

etwasmitbaum, 2021-06-22:
> @etwasmitbaum can you open a pull request?

I just did! This was my first time ever doing a pull request.

>Also, this is a workaround but the RTC is not supposed to have a drift like this, 2 minutes a day is an error of 1388 ppm, while the DS3231M is supposed to have an accuracy of 5 ppm. There is something wrong with the RTC.

Thanks for the info. I contacted SQFMI about it.