sqfmi/Watchy #55: Instructions for update via bluetooth

derak-kilgo, 2021-06-15:
Sorry if this is a dumb question. I'm new to hackable hardware.

I can put watchy in to firmware update mode and connect it to a device, but I'm not sure what to do after that.
Being connected doesn't add any ports in the arduinio ide and the mode isn't mentioned in the documentation.
Can compiled sketches be upload to watchy via bluetooth?

sqfmi, 2021-06-15:
@derak-kilgo the compiled sketches are uploaded via USB though the Arduino. For BLE, currently we only support uploading our pre-compiled watchfaces by clicking on "Try It! (beta)" at https://watchy.sqfmi.com/watchfaces . You should use a browser that supports Web BLE e.g. Chrome

notmatt, 2021-06-19:
@sqfmi Thanks! I tried basically everything except watch faces on the site ("why would firmware be there? 🤣 ")

sqfmi, 2021-07-31:
Closing this issue, will fix bug in https://github.com/sqfmi/Watchy/issues/65

closed 2021-07-31 by sqfmi

escopecz, 2025-07-23:
There is no "Try it" button there anymore, right? Or am I blind?