sqfmi/Watchy #56: Imperial Temperature Units not Working

rothnic, 2021-06-15:
I added `#define TEMP_UNIT "imperial"` to `Watchy_7_SEG.cpp`, which results in the watchface showing fahrenheit instead of celsius. However, I still am seeing the same 22 degrees as before. I'm have watchy 1.2.6 installed.

I have tried using different API keys and different cities. I've also tried to check the API call manually via postman, and I do not see how it is displaying the current value it is.

sqfmi, 2021-06-15:
Check out this documentation by @tapecanvas https://github.com/tapecanvas/WatchyDefaultFahrenheit

#define TEMP_UNIT "imperial" should be added to ```config.h```, the call here forms the query: https://github.com/sqfmi/Watchy/blob/aabb888069a54fe299f068e4171917bd3f6d1c9c/src/Watchy.cpp#L615

closed 2021-06-15 by sqfmi

rothnic, 2021-06-15:
> Check out this documentation by @tapecanvas https://github.com/tapecanvas/WatchyDefaultFahrenheit
> 
> #define TEMP_UNIT "imperial" should be added to `config.h`, the call here forms the query:

Right, is that not what I essentially said I did? We need to edit the library sources config.h file to change the setting, instead of setting it in the source for the watchface?

sqfmi, 2021-06-15:
We'll probably move config.h up to the watch face level, I agree it makes sense to make it easier to customize these settings at the watch face level, instead of in the library

rothnic, 2021-06-15:
I only work with arduino and c/c++ stuff rarely, so it looks like i assumed i could "override" the define settings for watchy. After some reading, it seems not that straightforward. So, yeah there needs to be a better way of passing in or overriding those settings. Otherwise, would each update of watchy not wipe those changes out?