sqfmi/Watchy #57: How to use Watchy with Platformio

silmaril42, 2021-06-15:
I don't really understand the folder structure of this project.

The example watch faces obviously contain `ino` files meant for Arduino IDE - but you can't do much here. I don't even see an option to change the city.

In the `src` folder it's all `cpp` and  `h` files that look more like something I would like to meddle with :smiley:  And I even found the `config.h` where I can set all the user settings.

Has anyone got this to work in Platformio yet? I seem to be able to install the Watchy library, but there is no board type to select for it.

I'd really love to use Platformio for programming this gadget and I'd appreciate any help someone with more knowledge of this could provide.

etwasmitbaum, 2021-06-16:
I converted the Watchy project to an PlatformIO project. Here is the .zip file of it:
[WatchyPlatformIO.zip](https://github.com/sqfmi/Watchy/files/6664949/WatchyPlatformIO.zip)
I manually included the Watchy library from source. If you want to use the precompiled like in the Arduino IDE just use this in the `platformio.ino`:
```
lib_deps = 
	sqfmi/Watchy @ ^1.2.6
        adafruit/Adafruit BusIO@^1.7.5
```

Professr, 2021-06-17:
You may also need to include this line in your platformio.ini:

`board_build.partitions = min_spiffs.csv`

For the larger demos, the compiled size will be too large otherwise.

silmaril42, 2021-06-17:
Great, thank you very much!

Now I am really looking forward to the weekend, when I hope to find some time to meddle with this. :+1:

adamm, 2021-06-19:
> The example watch faces obviously contain ino files meant for Arduino IDE - but you can't do much here. I don't even see an option to change the city.

How do I set the city/temp units?  This was my first question as well.  I submitted PR https://github.com/sqfmi/Watchy/pull/62 as a possible solution.

Obviously I know that I didn't purchase a fitbit here, as well the sqfmi developers have been focusing on the hardware production and not the software, but a little bit of instructions or comments in the examples would go a long way.   Even just as a hint as to where to go next.

etwasmitbaum, 2021-06-19:
@adamm If you are on Windows go to `C:\Users\[your username]\Documents\Arduino\libraries\Watchy\src` in there is a file called `config.h`. There are following lines:
```
27 //weather api
28 #define CITY_NAME "NEW+YORK" //if your city name has a space, replace with '+'
29 #define COUNTRY_CODE "US"
30 #define OPENWEATHERMAP_APIKEY "f058fe1cad2afe8e2ddc5d063a64cecb" //use your own API key :)
31 #define OPENWEATHERMAP_URL "http://api.openweathermap.org/data/2.5/weather?q="
32 #define TEMP_UNIT "metric" //use "imperial" for Fahrenheit"
33 #define WEATHER_UPDATE_INTERVAL 30 //minutes
```
In those lines you can set your location and temp units.

charles-haynes, 2021-07-30:
be a little careful. That platformio project specifies `board_build.flash_mode = qio` but the board description for pico32 says you should use `board_build.flash_mode = dio` I just got bitten by that yesterday. Worked fine for quite a while then mysteriously stopped working till I changed to dio mode.

etwasmitbaum, 2021-07-30:
> you should use board_build.flash_mode = dio 

Thanks for info. Wasn't aware of that, I just copy pasted what I used in Arduino.

I updated my comment above with the changed flash mode.

charles-haynes, 2021-08-01:
Actually, that's the default for `board = pico32` so you should probably just delete the `board_build.flash_mode = ...` line entirely. Most boards will specify their preferred flash mode.

BTW - thanks for posting that zip, I used it to get started and it was very helpful. I only found the qio/dio problem because after many days of working, suddenly uploads completed but failed to boot.

etwasmitbaum, 2021-08-01:
Just updated my comment from above again and removed the line.

I am glad to I could help you with my zip :)

brianmay, 2021-08-04:
It would be really good if somebody could supply a PR that gives a working platformio setup for the examples. I think you need:

* Move source files to src directory. This seems to be required for platformio. Hopefully it won't break arduino builds(?).
* Fix the path to Watchy source. Relative paths seem to be OK, so `../../..` is fine.
* platformio.ini file from above zip file.

Seems to work for me. So far.

Bertruss, 2022-07-16:
Yeah, I've been having a lot of issues getting this working. This is the first time I've used platformIO so I'm a bit unfamiliar with the configuration process as is though.

sommerper, 2023-04-20:
I'm trying to get the updated example from [above](https://github.com/sqfmi/Watchy/issues/57#issuecomment-862596850) to work but get this error:

```
...
src/Watchy_7_SEG.cpp: In constructor 'Watchy7SEG::Watchy7SEG()':
src/Watchy_7_SEG.cpp:14:24: error: no matching function for call to 'Watchy::Watchy()'
 Watchy7SEG::Watchy7SEG(){} //constructor
                        ^
In file included from src/Watchy_7_SEG.h:4,
                 from src/Watchy_7_SEG.cpp:1:
.pio/libdeps/pico32/Watchy/src/Watchy.h:52:12: note: candidate: 'Watchy::Watchy(const watchySettings&)'
   explicit Watchy(const watchySettings &s) : settings(s) {} // constructor
            ^~~~~~
.pio/libdeps/pico32/Watchy/src/Watchy.h:52:12: note:   candidate expects 1 argument, 0 provided
.pio/libdeps/pico32/Watchy/src/Watchy.h:44:7: note: candidate: 'Watchy::Watchy(const Watchy&)'
 class Watchy {
       ^~~~~~
.pio/libdeps/pico32/Watchy/src/Watchy.h:44:7: note:   candidate expects 1 argument, 0 provided
.pio/libdeps/pico32/Watchy/src/Watchy.h:44:7: note: candidate: 'Watchy::Watchy(Watchy&&)'
.pio/libdeps/pico32/Watchy/src/Watchy.h:44:7: note:   candidate expects 1 argument, 0 provided
Compiling .pio/build/pico32/libd61/Adafruit GFX Library/Adafruit_GFX.cpp.o
*** [.pio/build/pico32/src/Watchy_7_SEG.cpp.o] Error 1
======================================= [FAILED] Took 11.14 seconds =======================================
```

My `platform.ini` is as follows:
```
[env:pico32]
platform = espressif32
board = pico32
framework = arduino
monitor_speed = 921600
board_build.f_cpu = 240000000L
board_build.f_flash = 80000000L
build_flags = -DCORE_DEBUG_LEVEL=0
board_build.partitions = min_spiffs.csv
lib_deps = 
	adafruit/Adafruit GFX Library@^1.10.10
	arduino-libraries/Arduino_JSON@^0.1.0
	jchristensen/DS3232RTC
	zinggjm/GxEPD2@1.5.1
	https://github.com/tzapu/WiFiManager.git
	adafruit/Adafruit BusIO@^1.8.2
	sqfmi/Watchy@^1.4.6
```

So basically I'm importing `sqfmi/Watchy@^1.4.6` instead of the included watchy lib. But that one also didn't work which is why I'm trying this instead.

Is there maybe an updated working project using PlatformIO?

Thanks!

sommerper, 2023-04-20:
Never mind. I managed to fix it by deleting everything in the `src` folder and copying `Watchy/.pio/libdeps/pico32/Watchy/examples/WatchFaces/7_SEG` to `src`. Then renamed the `.ino` to `main.cpp`.

After pressing upload the project compiles and is uploaded to the watch.

joeleonjr, 2023-08-30:
Hi! I'm a security researcher @TruffleSecurity. I found a valid API key leaked in a comment in this Issue. I highly recommend rotating the API key immediately (not just deleting the comment). Here's a blog post I wrote on remediating these types of vulnerabilities: https://trufflesecurity.com/blog/remediate-leaked-api-keys-with-key-rotation.<br><br>If you can't find the leaked API key, you can use the following [TruffleHog](https://github.com/trufflesecurity/trufflehog) command to automatically scan for it:<br>`trufflehog github --repo https://github.com/<your_username>/<your_reponame> --issue-comments --only-verified`<br><br>If it's not clear where/what the security issue is, please let me know. I'd be happy to help. Please note that the API key might be in the past edit of a comment.<br><br><br><br>(cc @etwasmitbaum)

etwasmitbaum, 2023-09-03:
Hello @joeleonjr,
thank you for making me aware!
The API, key you are referring to, is the default API key from the Watchy source code [here](https://github.com/sqfmi/Watchy/blob/342eb48a497eb3dcb1ec03c5c156937fdec07ba9/examples/WatchFaces/7_SEG/settings.h#L6).
So it already was and still is publicly available. But thank you again for making aware.

DerLotke, 2024-08-17:
I also tried to compile my project with PlatformIO and had to find out there exists no solution in this thread to build the recent version on current hardware revisions (aside from [what is available on the PlatformIO registry](https://registry.platformio.org/libraries/sqfmi/Watchy)).

That is why I set up a wrapper-repository that fixes this issue [here](https://github.com/DerLotke/Watchy-PIO).

It has everything needed to get started quickly and also features a configure-script that allows you to choose your preferred watchface.