sqfmi/Watchy #63: Experiences, questions and documentation improvement ideas for beginners

mikkjant, 2021-06-18:
Beginner experiences so far. Blindly go where someone has gone before.

1. Opened the box and started to assembly according instructions https://watchy.sqfmi.com/docs/getting-started.
- I finished the assembly and checked that watch face was on. I did not find any documentation what button is what so I just pressed them all. Managed to find menu and tried battery information and it stuck there. No response from any button. Whatchy is now stuck. Ok - I guess I need to re-flash it.

2. https://watchy.sqfmi.com/docs/getting-started#arduino-setup
- Then I started with arduino-setup. Could not get it to work with Mac but with linux it was ok.
- stumbled on 
	- https://playground.arduino.cc/Linux/All/#Permission 
	- "no-serial module" https://github.com/espressif/arduino-esp32/issues/13 solved with ```pip install pyserial```

3. https://playground.arduino.cc/Linux/All/#Permission
- make sure USB cable can handle data. 
- Compile & Upload with these board settings - were missing from above instruction which I found from here: https://github.com/sqfmi/Watchy#setup

4. Now I could upload 7_SEG watch face again - and get it working

Watchy code
Watch face examples were there in Ide but it took some time for Arduino beginner to find where Watchy code is. 
$ ~/Documents/Arduino/libraries/Watchy
For example: config.h - change Weather location here 

Default Watch face - instruction
- Button details - which button does what. LeftUp - back, LeftDown - menu/action, RightUp - Up, RightDown -Down
- Manu item functions
	- Battery Check
		- will stuck if used
	- Vibrate motor
	- Show Accellerometer
	- Set Time
		- works
	- Setup Wifi
		- works if finished other vice stuck
		- connect to your Watchy Wifi AP and follow instructions
	- Update firmware - no idea how to do this or what this does?
		- will stuck

Common problems and solutions
- Stuck to menu
	- personally I can not much use menu without it getting stuck
	https://github.com/sqfmi/Watchy/issues/49
	- I have Aluminum Case - I have not wanted to open it again as don't want to crack the screen.
	- workaround - upload 7_SEG again	
	- solution?
- Check out other issues if problems: https://github.com/sqfmi/Watchy/issues

Questions:
- Does Ide watch face upload update firmware?
- how to update firmware?
- BLE information?
- any good links to Watchy relevant Arduino beginner documentation?

Possibly this experience can be used to improve documentation and after learning something -> I guess Arduino beginner documentation PR's and some other relevant documentation PR's are ok to do? :)

aravindet, 2021-06-19:
Just wanted to chime in that as a fellow beginner, I found @mikkjant's list above immensely helpful. I spent a great deal of time on:
- trying to use a USB cable that doesn't support data. Yes, I'm that dumb, but I have a feeling I am not the only one :joy: 
- pyserial not found
- https://playground.arduino.cc/Linux/All/#Permission

SailingGreg, 2021-06-22:
@mikkjant I found that you can update the firmware via Bluetooth in Chrome

Go to https://watchy.sqfmi.com/watchfaces and select the watch face you want to download - this will open a popup and look to find the Watchy
Go menu/update firmware and select update which will allow you to pair - confirm on the laptop
The download will start but it takes a while as it around 2k/second and it around 1.7M
The device will reboot

jazzgroove, 2021-06-22:
> Just wanted to chime in that as a fellow beginner, I found @mikkjant's list above immensely helpful. I spent a great deal of time on:
>
> * trying to use a USB cable that doesn't support data. Yes, I'm [...]

This was very helpful. After determining that the USB tty was in a certain group, adding my user solved the connection problems. One pro-tip on Linux: after adding yourself to a new group just sudo to your own user to refresh the group.

seclorum, 2021-08-03:
One thing that I found a bit dumb was that the protective sleeve that covers the e-ink display gets splotchy and doesn't quite attach itself to the display with the glue evenly.  I have had to really squish the cover onto the display and even then the blotches don't disappear - tried smoothing them out gently with a flat edge.  One bit of advice might be to ensure you don't get fingerprints on that protective cover while you're applying it ..

theposhbloke, 2021-08-14:
Had the same problem with screen protector as @seclorum. Also watch keeps terrible time: it has lost 4 minutes in two days use. I know the watch is designed to be hackable, but if it can't do the basic function of keeping reasonable time then it is pretty useless. I don't feel enthusiastic about exploring it further because of this.