sqfmi/Watchy #65: Firmware update can't be cancelled/exited

notmatt, 2021-06-19:
![watchy screenshot][watchy]{width=40px height=400px}

[watchy]: https://user-images.githubusercontent.com/92480/122651156-c3744100-d0eb-11eb-8610-53e03d5d5962.png

notmatt, 2021-06-19:
Uploading one of the available watch faces at https://watchy.sqfmi.com/watchfaces did work for me and (obviously?) exited the update mode.

waynew, 2021-06-28:
The way I got out of this was actually by connecting to the device via my phones BT, then I unpaired, and the watchy failed and returned back to the main screen.

sigurdlasa, 2021-07-01:
> Uploading one of the available watch faces at https://watchy.sqfmi.com/watchfaces did work for me and (obviously?) exited the update mode.

I got trapped in this problem today and uploaded a watchface to exit, all the while hoping I didn't brick it.

hugobuddel, 2021-07-16:
I couldn't manage to send a watchface, so I unpaired which also exited the bluetooth update screen. But then it was stuck on the menu, so needed to remove the battery anyway.

lightnin, 2021-07-30:
This seems like a pretty bad bug if there is no way to recover from this mode.

seclorum, 2021-08-03:
Just wanted to report in that I was stuck with this conundrum today - until I used the right browser.  With Google Chrome I was able to make the BLE connection on my MacBook Pro, upload a new watch face, and reset the watch without having to disassemble/reconnect the battery.

tlhintoq, 2022-07-30:
29aug2022
Windows10 laptop (MSI GS76 Stealth, latest everything)

Google Chrome will *attempt* and act like its updloading one of the "try it now (beta)" faces per #55, and will reboot the watch.
But the new face is not installed.

FYI
This also does not resolve the issues with the 7-seg display failing to report the battery level correctly.
https://github.com/sqfmi/Watchy/issues/153

crossplatformdev, 2024-07-05:
Watchy.cpp:1059
+++ if(digitalRead(BACK_BTN_PIN) == ACTIVE_LOW) break;