sqfmi/Watchy #66: Aluminum top case doesn't _quite_ fit PCB (easy fix though)

notmatt, 2021-06-19:
Leaving this here for anyone who runs into a similar problem, or maybe an addition to the FAQ or troubleshooting sections. My top case didn't _quite_ fit the PCB, and the problem ended up being the leftover tabs from PCB manufacture. There's four of them, two on each of the right and left sides.

[image]

This is an easy if delicate fix: the components do not have a lot of clearance from the nominal edge of the PCB! But one or two light strokes with a small file, taking care to keep it perpendicular to the PCB, and problem solved:

[image]

notmatt, 2021-06-19:
Anticipating a couple obvious questions:

No, it wouldn't force in, at least not with the amount of pressure I'm comfortable exerting on a PCB.

No, absolutely zero wobble after this. The manufacturing tolerances on the case and PCB are _tight_ (modulus those tabs, but that's hard to avoid).

SnoopJeDi, 2021-06-21:
Confirming that I (and some folks on Discord) had the same issue, it appears the tolerances sent to the shop were indeed very tight. I knocked down the tabs with some 150 grit sandpaper and the PCB settled down nice and snug after that.

mpboddie, 2021-07-16:
I had the same issue but I also had to lightly sand some extra bits of aluminum from the inside of the aluminum case by the USB port on the top half. Once I got rid of the extra pcb tabs and rough aluminum edge, it fit well.

GuruSR, 2021-07-30:
Just a suggestion:

For the high peaks, a hobby knife to take some of those off, the other part, a decent emery board (yes, for nails), fits perfectly in those areas and you won't run the risk of damaging anything nearby because there is no grit on the sides.

Mine went in fine doing what I suggested.

bushmango, 2022-05-12:
Thanks! Filing these down with a metal file worked for me