sqfmi/Watchy #72: Blank screen after assembly and power on

gkzsolt, 2021-07-04:
My watch does not show anything on the screen (it is all blank, white-grey). When I first assembled, I forced the FPC connector, I didn't know that I should pull it first (this was not written on the first instructions, just "Connect the screen to the FPC connector". Also the video seems to be an accelerated recording, nothing important can be discerned. The second video is much better, and the instructions also say "Pull lightly to unlock the FFC connector". Sorry for me being so lame ;()

Anyway, the screen connector seems OK now, but it shows nothing: connecting power cleared it up. I connected Watchy to an Arduino IDE and uploaded some sketch (apparently OK, the last log says "Hard resetting via RTS pin..."), but no avail, screen blank. The buttons don't do anything. Is there a way to debug and check if the screen is connected, ...?

gkzsolt, 2021-07-27:
It seems that nobody from sqfmi is watching this list, or does not care. I know that this is not a commercial product, but I am a (sad) project backer. So I invested my money in vain...
Can somebody help? Is there a way to check if the screen is operational at all?
Thank you,

RyanH54, 2021-07-28:
Yeah unfortunately I don't think they really care. I'm trying to get my refund through CrowdSupply but they say SQFMI told them they already responded to my request when in fact they've been completely silent except for giving me and others a link to a Google Form on June 14 that I immediately filled out and never heard back from. I tried giving CrowdSupply the link to the issue page here on GitHub which clearly shows that nobody has been helped by SQFMI, but I don't know if they're reading that or ignoring ti.

gkzsolt, 2021-07-28:
You at least succeeded to contact them, @RyanH54 . How was it possible?

RyanH54, 2021-07-28:
> You at least succeeded to contact them, @RyanH54 . How was it possible?

Only interaction with SQFMI was [through this linked issue thread ](https://github.com/sqfmi/Watchy/issues/39#issuecomment-885237841)where they gave us all a Google Form to fill out, but as far as I can tell they never responded to anyone after filling out the thread. I went to CrowdSupply's site last Tuesday 7/20 and filed a ticket which they responded to today.