sqfmi/Watchy #73: Can't seem to connect to WiFi

ScienceMarc, 2021-07-12:
I just got my Watchy and it seems like the wifi feature does not work. I go through the menu and connect to the AP and enter the SSID and password and the watch responds with the message "Connected to [my network name]". I then press the top left button to go back but the watch face indicates that I am not connected to the internet (the wifi symbol on the default watch face is crossed out) and the weather doesn't seem to update.

My router's device list includes esp32-arduino so it seems to connect just fine. The router lists the device as being disconnected but from what I understand that is to be expected because the watch disables its radios to save battery life.

I have tried to restart the device through the web interface and I have tried erasing the WiFi config to no avail. I have also tried to connect to my phone's hotspot but that didn't seem to change anything.

sergneri, 2021-07-15:
After I simply loaded the screen 7_SEG into my Arduino editor as a connectivty test, compiled it then uploaded to watchy, the internet icon started working. Before, it was as you describe above.

ScienceMarc, 2021-07-19:
That seems to have worked. Thanks. It's a bit strange cause I had loaded that watch face multiple times while troubleshooting but doing it again today it decided to connect to the wifi.

closed 2021-07-19 by ScienceMarc