sqfmi/Watchy #74: Watchy's RTC is slow

slugh23, 2021-07-17:
According to the data sheet for the DS3231MZ, the accuracy of the RTC should be:

> Timekeeping Accuracy ±5ppm (±0.432 Second/Day)
> from -45°C to +85°C

I received my watchy three days ago (Wednesday) and set the time to within ±1 minute.  Watchy has been sitting on my desk since it arrived.  Within a day it lost 1-2 minutes.  I reset the time to ±1 minute.  Today--two days after the last adjustment--watchy is now 3-4 minutes slow.

putz612, 2021-07-17:
Same issue for me, received it this week and have lost a few minutes every day.

vanEgmond, 2021-07-20:
Probably related: #40

slugh23, 2021-07-22:
Seems to be.  Mine is marked 1611A3 597AD.  Now just wondering what the remedy is.

vanEgmond, 2021-07-22:
Contact sqfmi on Discord or write them an email, they are currently figuring out logistics as it seems like a bad batch.
1611A3 597AD is one of the serials that keeps popping up.

slugh23, 2021-07-22:
Thank you very much!

closed 2021-07-22 by slugh23