sqfmi/Watchy #76: Wifi Credentials in Config.h

Kongduino, 2021-07-19:
In https://watchy.sqfmi.com/docs/libs#wifimanager it is stated:

"Alternatively, you can also hard code your [wifi] credentials in the config.h."

How? What should be used.

https://github.com/sqfmi/Watchy/blob/master/src/config.h#L36 has a `WIFI_AP_SSID`, but that's for wifi setup. What should the defines be for our own wifi credentials?

sqfmi, 2021-07-25:
You can pass the credentials directly inline here 

https://github.com/sqfmi/Watchy/blob/aabb888069a54fe299f068e4171917bd3f6d1c9c/src/Watchy.cpp#L822

or set them in ```config.h``` as a define/const and pass it in the same way

brianmay, 2021-08-03:
How do you recompile and use this file? The instructions say to install the Watchy library, which I think means it will download these files again instead of using the local copy that you just changed.

Or maybe this is where I should be using PlatformIO?

brianmay, 2021-08-04:
Oh, I think I need a symlink. At least on Linux. No idea on Windows (which I don't use).

```
rm -rf ~/Arduino/libraries/Watchy
ln -sf ~/tree/3rdparty/Watchy ~/Arduino/libraries/Watchy
```

Seems to work now.

brianmay, 2021-08-04:
Also for reference, I tried PlatformIO, but couldn't get the dependencies correct, so ended up giving up on that for now.

sergneri, 2021-08-10:
at the bottom of config.h  - add these lines:

//WiFi setup for my network
#define WiFi_SSID "myssid"
#define WiFi_PASS "passwordformyssid"

save and do a compile. This compiled fine and appears to work fine on 7_SEG and threw a warning on StarryHorizon face but worked.

closed 2022-01-20 by sqfmi