sqfmi/Watchy #77: Fails to increment time, per startup instructions

billsalt, 2021-07-20:
Hi,
  I followed the setup instructions: attached display, attached battery, plugged into USB.  The watchy is responsive as documented (buttons, display changing) and I can scroll through the setup menu.  Exiting and waited a few minutes, the time never increments.  

  Do I have a bad unit, or are the instructions incorrect?

Thanks!

vanEgmond, 2021-07-20:
Possibly/probably related: #40

billsalt, 2021-07-20:
Thank you @vanEgmond, I agree (somehow missed that).  It seems several of us are having issues and I'd have to assum there is either a design issue or an assembly issue.  

What I don't see is any definitive response from @sqfmi (unless I'm missing that too).  Would you folks please comment on these to let us know what is going on?  I'd be happy to send mine back since it is not functional per my needs; send me an RMA and I'll do so.  I'd certainly expect a functional unit in return.

I appreciate that these issues can arise for a new product and company; hopefully they are efficiently resolved.

Thank you.