sqfmi/Watchy #79: How on Earth do you configure WiFi/Bluetooth/etc?

sterlingphoenix, 2021-07-29:
I'll admit that I'm absolutely not an arduino expert, so I'm stumbling around in the dark here. 

I have no idea how to configure wifi or bluetooth. I was able to load a new watch face, and through that I found SOME wifi settings in config.h, but I have no idea how to make the Arduino IDE actually, you know, do anything with that. I see that this project uses WiFiManager but that's also not very helpful. 

There are a lot of other things hardcoded into config.h that would be nice to change. 

I do see some issues asking similar questions, but no actual solutions, so... can we get some docs on this?

shtrom, 2021-08-03:
When you press the menu button (bottom left), you get to a menu which has a “Setup WiFi” option. Selecting this will drop the watch into an AP mode (Watchy AP), that you can connect to. It runs a lightweight web server where you can choose one of the available access points, and enter the key, to connect to.

OTOH, I don't think there is actually a functional Bluetooth support at the moment.

joetechie, 2021-08-15:
This is likely obvious to others, but it wasn't to me -- you need to use another device, like a phone, to connect to the wifi signal the Watchy is sending. You do that in the settings of the phone, choosing the Watchy as the wifi. It will pop up a sign-on sheet, where you can enter the SSID and password you want the Watchy to connect to.

ohmygaugh-crypto, 2024-09-08:
For clarity this is the process:

0. Have a Computer / Phone at hand and open your wifi settings (You'll be looking for the "Watchy AP" wifi signal to appear)
1. On the Watch. Press "setup wifi" option. It will briefly transmit the network signal. On your Computer/phone be scanning your screen for the "Watch AP" network to appear within the wifi settings.
2. You'll be redirected to a webpage that allows 4 options. 

- Configure WiFi (click this one)
- Info
- Exit
- Update

3. For SSID field state the network name you want the watch to establish connection with. For Password you will use the same password that connects to that associated wifi you have at your home/mobile hotpost.
4. Wait a second for the watch to connect. It will. then Enjoy. 

 *note* on my Android I had to 1st add a network called "Watch AP" within the wifi settings before pressing "setup wifi" on the watchy. I then used my hotspot device name as the SSID and the associated password, but I'm sure you could also tether it to your home router as well.