sqfmi/Watchy #80: CNC case + Plastic buttons

GuruSR, 2021-07-29:
Just actually got the Watchy today, cleaned the circuit board edges so it sits perfectly flush inside the CNC lid of the case, the same can't be said about the plastic buttons, upon attempting to screw down the case, the lid starts to buckle as the plastic buttons are too tall.

Those blessed with a small bench vise, can put all 4 buttons in and lightly sand around the edges, this will give the buttons the needed relief and stop the CNC lid from buckling.

If you don't have a small bench vise capable of holding the buttons without damaging them, use gripping tweezers and a very mild nail file (as fine a grit as you can get it) and lightly sand it around the entire edges where the button sits in the mount.

For those having issues with the plastic case, the same method may make the buttons move better.  Maybe in the future, custom  machined buttons for the CNC case would be great.

Harry314, 2021-07-29:
My CNC case came with aluminium CNC buttons.
They were in the pack with the Allen key/Hex key.

GuruSR, 2021-07-30:
Ah, I'll have to double check that, it didn't seem to have them.  Thanks, I'll look.

closed 2021-07-30 by GuruSR

synthead, 2022-05-12:
I also experienced this with the factory buttons with [this aluminum case](https://shop.sqfmi.com/products/gb-watchy-cnc-anodized-aluminum-case).  I had to file around the "rectangular" parts of the buttons and a little in the button slots for the buttons to not be bound to the case after tightening the screws.