sqfmi/Watchy #81: Link to discord server on homepage expired

lightnin, 2021-07-30:
The link to the discord server on the FAQ page here:
https://watchy.sqfmi.com/docs/faqs

Is expired, and results in this  error:

[image]

It seems like you could use some better support channels, before the returns start flowing back.

lightnin, 2021-07-30:
Note that searching for "SQFMI discord server" doesn't yield any useful results, so AFAIK the only way to get support is to file an issue on github.

sqfmi, 2021-07-30:
You can try to Discord link on our homepage https://discord.gg/ZXDegGV8E7 , that link appears to have expired and needs to be updated

closed 2021-07-30 by sqfmi