sqfmi/Watchy #83: Are all Arduino projects this rough?

mikehihz, 2021-07-30:
I was really looking forward to having this be my first Arduino adventure, but even though assembly went fine, Watchy booted up but the two right buttons were non-responsive and the bottom right one popped off when I was making sure the buttons worked out of the case. https://github.com/sqfmi/Watchy/issues/39

Even so, hitting menu button to set time would give me garbage to the screen, uploading firmware never worked, I couldn't connect through WiFi, and now Watchy is locked up again. 

This is pretty rough stuff and I opened a ticket with SQFMI, but are all Arduino/open source projects in this rough state when they ship hardware? Were my expectations of beta-quality software for an MVP too high?

sterlingphoenix, 2021-08-01:
No they're not. But most of them aren't an open-source project meant for people who absolutely want to hack the device. 

For reference, I'm struggling with this one, too. Never had a _smidgen_ of problems with, say, adafruit products.

seclorum, 2021-08-03:
Had my Watchy for a day, and I have to say I haven't run into any of these sorts of issues yet - seems to work pretty smoothly, and its nice to be able to get the sources for things out of the box, build, and upload .. about the only hiccup I had was when I selected "Update Firmware" from the menu, and couldn't get out of the screen - then I realized that I should go to the https://watchy.sqfmi.com/watchfaces/ site and try to upload a new face and see if that reset things - this didn't work until I used Google Chrome, which has access to BLE services where other browsers don't.

All things considered, its a fairly normal Arduino project.  There are definitely hoops to jump through if you're new to all this - the board manager setup, etc.  But if you're a half-competent developer, none of this should get in your way.

brianmay, 2021-08-04:
I think this software project is relatively immature. There are a number of PRs open to fix various problems. Hopefully these will eventually get merged and gradually the quality will improve.

closed 2021-12-31 by sqfmi