sqfmi/Watchy #88: Watch Face Doesn't Change After Successful Sketch Upload

tex792, 2021-08-01:
The watch face does not change from the default screen that it ships with. When I upload a sketch, it seems to upload successfully and ends with "Done uploading." in the Arduino IDE.

Could a crack on the back of the screen possibly affect the entire display? When I very first opened the package, the screen slipped out and landed on the floor. It has a visible crack on the back, but the front looks ok.

sqfmi, 2021-08-02:
A crack would probably affect the screen from refreshing, also make sure the ribbon cable is properly connected. You can purchase a replacement screen here https://www.aliexpress.com/item/33044560386.html. The LED indicates charge status and is not connected to any GPIOs.

tex792, 2021-08-03:
Thank you! I checked in the Discord chat as well and folks seemed to think the same thing. I went ahead and ordered a few of the replacement screens from AliExpress. Also that's good to know regarding the onboard LED; I wasn't aware of that.

closed 2021-08-03 by tex792