sqfmi/Watchy #93: WiFi setup error: "Setup failed and timed out"

naikymen, 2021-08-05:
I get this message immediately selecting `Setup Wifi` at the main menu:

```
Setup failed & timed out

naikymen, 2021-08-05:
Solved it by compiling the example and uploading it from another machine.

The problem is with my main linux installation; Arch linux has issues compiling arduino programs for whatever reason.

Just FYI, what i did was:

1. Assemble the Watchy and realize that the buttons were not working.
2. The WiFiManager library did not compile on my system, unless I used a different branch:
  * https://github.com/tzapu/WiFiManager/tree/esp32s2
  * https://github.com/tzapu/WiFiManager/issues/1246
 3. So i used the other branch, and uploaded the code.
 4. The buttons were still not working, and then i found that they are not working well: https://github.com/sqfmi/Watchy/issues/49
 5. Then i tried setting up wifi and got the `Setup failed & timed out!` error.
 6. Using the main WiFiManager branch was not an option on my main PC, so i used an old chromebook with GalliumOS and everything turned out fine.

So far so good! :)

closed 2021-08-05 by naikymen

rlhelinski, 2021-09-10:
I can confirm that the examples for Watchy, which use WiFiManager, do not compile version 2.0.0 of the esp32 board package, but they do with version 1.0.6.  The API must have changed. To help others who are searching for help with the error, here is what I got from the Arduino IDE:
```
/home/ryan/Arduino/libraries/WiFiManager/WiFiManager.cpp: In member function 'void WiFiManager::WiFi_autoReconnect()':
/home/ryan/Arduino/libraries/WiFiManager/WiFiManager.cpp:2961:79: error: no matching function for call to 'WiFiClass::onEvent(std::_Bind_helper<false, void (WiFiManager::*)(arduino_event_id_t, system_event_info_t), WiFiManager*, const std::_Placeholder<1>&, const std::_Placeholder<2>&>::type)'
       wm_event_id = WiFi.onEvent(std::bind(&WiFiManager::WiFiEvent,this,_1,_2));
                                                                               ^
```