sqfmi/Watchy #102: BLE Battery Duration

HerrAugust, 2021-11-09:
Can anyone tell me about the estimated battery duration when the smartwatch has:

    screen always turned off (or turned on only for a new notification)
    BLE connected all the day to an Android v8 smartphone (to received message and email notifications with GadgetBridget)
    WiFi always turned off

Thank you a lot!

PS: it seems that this smartwatch is not that actively developed anymore?

LeonMatthes, 2021-11-11:
Hi, the screen actually doesn't use any energy, as long as it's not updating, so you don't have to ever "turn it off". That's the awesomeness of e-ink. You can have it display whatever you want and it won't use any energy, as long as its a still image :)

This does mean however, the microcontroller in Watchy (ESP32) uses the bulk of energy of the watch.
With BLE active, it has a power consumption of around 80mA. With the standard 200mAh battery, it will last ~2 1/2 hours...
With WiFi active, less than 2 hours.

If you manage it carefully, keep the microcontroller in deep sleep most of the time, and only connect to your phone every minute, you should be able to get it to ~2 days of battery life.

Dunno about how active development on Github is, but people seem relatively active on Discord.

HerrAugust, 2021-11-11:
Hello, thank you for the answer! I have a similar experience with another smartwatch, Lilygo TTgo watch V3, and also there with only Bluetooth turned on (and screen and wifi off) the battery lasts the same time.

But they did the trick with deep state and other things, as you can see here:

https://github.com/sharandac/My-TTGO-Watch/issues/316

So my question is if this firmware supports notifications from GadgetBridge in Android? And in your experience, the smartwatch resists for 2 days with BLE turned on in this firmware?

https://github.com/sqfmi/Watchy/issues/102#issuecomment-966494253