sqfmi/Watchy #103: unable to set time

trondhindenes, 2021-11-16:
my Watchy is, unfortunately, unable to tell me the time. 

I can use the menu system to set the date/time, but it doesn't seem to "save" anything - when I'm back to the watch face the clock is always "00:00".

zouloux, 2021-11-17:
Hello, same issue here

zouloux, 2021-11-17:
Tested this : https://github.com/JChristensen/DS3232RTC
It seems to be working, I see time updated every seconds in serial output.
I'm unable to set time, battery voltage is crazy off (it shows 0.28 volts), temperature is off (30°c or 100°c).
Time was working before I uploaded a new firmware (testing with 7 seg freshly downloaded from github + followed instructions to config Arduino)
Thanks

chiefdaft, 2021-11-17:
Same here. Uploaded the Watchy_7_Seg example with my own Open Weather Map API key. Time is updated every minute so the clock is running, but I can't store the actual time via the Watchy settings menu, neither can I store the WiFi settings via the Watchy AP. The webpage comes up, you can select a SSID and enter a password, but the Watchy is not able to store it.

castleberrysam, 2021-11-18:
Hello all, I am having the same issue.

Upon closer inspection, it looks like I received a different revision of the device, which has not been documented anywhere as far as I can tell. See the below photo.
[image]
It looks like the DS3231 RTC chip has been replaced with the PCF8563 (also an RTC chip). Setting the time does not work with the current code since the chip interfaces are not the same. I suspect that the incorrect battery voltage readings could also be due to a hardware change, but I haven't confirmed that yet.

My device shipped with fully functional software, and only stopped working once I tried to use the code from this repo, so it seems that someone out there has a copy of the code working with the new chip. Since I'm not sure if/when the repo might be updated to use the new chip, I will also probably try to update the code myself in the next few days.

zouloux, 2021-11-18:
Nice find @castleberrysam ! Will check on mine if this is the same chip. It's cool we are not alone, but it's kind of weird that nobody at @sqfmi seems aware of this ? The Watchy arduino should be updated to support PCF8563. Software shipped with the package was working well also with mine so it should be nothing to update the new software.

chiefdaft, 2021-11-18:
I did a check on my board, and guess what?  Obviously SQFMI changed the flunky batch of Maxim Integrated DS3232 RTC chips with another RTC chip set.  The replacement chip is a PCF8563 indeed. (https://www.nxp.com/docs/en/data-sheet/PCF8563.pdf) Someone wrote an example code on http://www.esp32learning.com/code/esp32-and-pcf8563-rtc-module-example.php with this chip, library and esp32. Hopefully we can find a solution for the new chip and the Watchy configuration.
[image]

chiefdaft, 2021-11-18:
The PCF8563 doesn't seem to have temperature sensor like the DS3232RTC. :-(

zouloux, 2021-11-18:
Not 100% sure but temperature seemed correct before I pushed a new firmware.

Dexdev08, 2021-11-18:
i just tried out my watchy received just today. didn't even get to test the software that came with it - i just uploaded the examples and I encountered the same - the time is stuck at 00:00

trondhindenes, 2021-11-18:
is anyone from the supplier monitoring these threads?

Dexdev08, 2021-11-18:
i got the original firmware back in discord.

trondhindenes, 2021-11-18:
@Dexdev08 oh nice - I'd like to try that as well. Do you have a link or something?

chiefdaft, 2021-11-18:
Help is under way! See thread on Discord: https://discord.com/channels/804832182006579270/804834557412900885/910362640386248704

brianmay, 2021-11-18:
Can somebody please provide a link to the original firmware? I thought I had a discord account, but doesn't appear to be letting me into that channel.

My watch got into a state where it will accept uploads fine, but there is no evidence of attempting to run the new code. Thiis is with code that was previously working. So it would be really be good if I could get "known good firmware" to prove that I am not doing something idiotic/wrong with the build/upload (I doubt it).

chiefdaft, 2021-11-18:
[watchy_new_rtc.zip](https://github.com/sqfmi/Watchy/files/7566665/watchy_new_rtc.zip)
Run the esptool with: esptool  write_flash 0x00000 watchy_new_rtc.bin
BTW, for the upload with esptool.py to load succesfully I had to remove the battery, and just before hitting enter on the command line attach the USB cable to the Watch
[image]
y

andrew-pa, 2021-11-18:
Do we have any idea when the source code will be available?

chiefdaft, 2021-11-18:
> Do we have any idea when the source code will be available?

according SQFMI on Discord it will be over the weekend, next week. There are still some bugs in it. Like the day and the date don't match. It will be pushed to GitHub

trondhindenes, 2021-11-18:
Thanks! I managed to flash the firmware (didnt need to reset the battery or anything), but now my problem is that the buttons don't work. The initial self-test, in which the buttons seems to be "mismatched, completes, but when the regular watch face shows, none of the buttons do anything.

castleberrysam, 2021-11-19:
The buttons are also mismatched for me. The buttons weren't working initially but they are working now, so maybe you can mess with it/reset it a few times and see if it starts working.

brianmay, 2021-11-20:
My tests seem to show that the DS3231 RTC isn't particularly accurate. The DS3231 seems to be loosing time[1]. Wonder if the PCF8563 is any better?

[1] My latest tests suggest it looses about 1 second every minute.

chiefdaft, 2021-11-20:
My Watchy still runs perfectly in sync with my laptop time after two days. I have an older Watchy with a broken DS3231 on board that runs it's time on the ESP32 rtc. This one drifts a few minutes per day. By the way, the new one still is not capable storing the WiFi settings. Let's see if that improves when the next version of source code is published.

dboitnot, 2021-11-20:
I was able to get the PCF chip to work using this library: https://github.com/orbitalair/Rtc_Pcf8563

Here's a quick PoC which sets the time then wakes up every minute or on a button press and reports the time from the RTC.

```c++
#include <Arduino.h>
#include <DS3232RTC.h>
#include "Rtc_Pcf8563.h"
#include "config.h"

#define SERIAL_BAUD 115200

RTC_DATA_ATTR bool state_initialized = false;

Rtc_Pcf8563 rtc;

void deep_sleep() {
  esp_sleep_enable_ext0_wakeup(RTC_PIN, 0);
  esp_sleep_enable_timer_wakeup(60000000);

  // enable deep sleep wake on button press
  esp_sleep_enable_ext1_wakeup(BTN_PIN_MASK, ESP_EXT1_WAKEUP_ANY_HIGH);
  esp_deep_sleep_start();
}

void init_state() {
  // Give the serial console time to connect
  delay(10000);

  // Initialize the realtime clock
  rtc.initClock();
  rtc.setDate(14, 6, 3, 0, 10);
  rtc.setTime(1, 15, 40);

  state_initialized = true;
}

void setup() {
  Serial.begin(SERIAL_BAUD);

  if (!state_initialized)
    init_state();

  Serial.print(rtc.formatTime());
  Serial.print("  ");
  Serial.print(rtc.formatDate());
  Serial.print("  0x");
  Serial.print(rtc.getStatus2(), HEX);
  Serial.print("\r\n");

  deep_sleep();
}

void loop() {}
```

sqfmi, 2021-11-22:
Hey all, just pushed the latest firmware to branch [pcf8563](https://github.com/sqfmi/Watchy/tree/pcf8563) , make sure you have https://github.com/orbitalair/Rtc_Pcf8563 installed as well. All watchfaces should be supported.

sqfmi, 2021-11-22:
Hey all, just pushed the latest firmware to branch [pcf8563](https://github.com/sqfmi/Watchy/tree/pcf8563) , make sure you have https://github.com/orbitalair/Rtc_Pcf8563 installed as well. All watchfaces should be supported. Eventually we'll merged it into master so that both RTCs are supported. Sorry for the wait!

epilys, 2021-11-22:
Confirming it works, thank you @sqfmi !

trondhindenes, 2021-11-27:
@sqfmi could you provide a bit more instructions? My watchy is still missing basic functionality like keeping the time, even with the new binary that was posted to Discord last week. I'm not familiar enough with the Arduino IDE to know how to use different branches in there.

I'm at the verge of requesting a refund, but I'd really like to try and make my watch work. Would very much like a proper "do A, B, then C" to try and troubleshoot my issue.

patvdleer, 2021-11-27:
in `platform.ini`

```ini
lib_deps =
    https://github.com/sqfmi/Watchy#pcf8563
    https://github.com/orbitalair/Rtc_Pcf8563
```

that should be it

chiefdaft, 2021-11-28:
> in `platform.ini`
> 
> ```ini
> lib_deps =
>     https://github.com/sqfmi/Watchy#pcf8563
>     https://github.com/orbitalair/Rtc_Pcf8563
> ```
> 
> that should be it

Great! A few days ago I compiled the new Watchy branch by copying my Rtc_pcf8563 libraries from the Ardiuno library into my local pio lib. That was working too, but a reference in the platformio.ini is better maintainable. Thanks for the tip.

sqfmi, 2021-11-29:
Just pushed v.1.2.8 with both RTCs support merged, you no longer require the PCF8563 branch. Install it with the library manager and it should be all good to go :)

trondhindenes, 2021-11-29:
Nice, thanks. I updated the library in Arduino, pulled down an updated "master" and installed the "basic" watch face. it works in that the time seems to increase, but buttons still don't work so I don't have a way of setting the time or getting into the menus.

sqfmi, 2021-11-30:
@trondhindenes join us on Discord and let's try to troubleshoot this issue, it should be unrelated to the RTC

epilys, 2021-11-30:
> @trondhindenes join us on Discord and let's try to troubleshoot this issue

Unrelated to the issue, but please consider implementing an IRC bridge or something. Not everyone wants to use Discord since it's proprietary 😢

closed 2021-12-12 by sqfmi