sqfmi/Watchy #104: Unable to compile

fabqu, 2021-11-22:
Hi!
I just got my Watchy, installed the libraries and tried to compile any example or watchy-face from GitHub - with the same result:
libraries\Watchy\src\Watchy.cpp: In member function 'float Watchy::getBatteryVoltage()':
libraries\Watchy\src\Watchy.cpp:652:40: error: 'analogReadMilliVolts' was not declared in this scope
     return analogReadMilliVolts(ADC_PIN) / 1000.0f * 2.0f;
Seems as if the function "analogReadMilliVolts" is not declared for some reason.

What exactly is happening there? Maybe a wrong or buggy library somewhere?

sqfmi, 2021-11-29:
Hello @fabqu, what version of ESP32 Arduino Core do you have? It should be in there:

https://github.com/espressif/arduino-esp32/blob/399f4ecbb3a4cef21e2bffa37adb6190356dfb76/cores/esp32/esp32-hal-adc.h#L45

fabqu, 2021-12-08:
Thanks a lot. Updating my core was the correct solution!

closed 2021-12-08 by fabqu