sqfmi/Watchy #106: Watchy buttons not working, time stuck at 00:00

MoMMUU, 2021-11-27:
Windows, Arduino 1.8.16

The watch face is working, but the buttons won't do anything, the time is stuck to 00:00, the battery indicator says it's empty, the temperature is 125, and there are 2 "ERR" markings under the 2nd 0 in the time.

I didn't change anything with the watch face code, so I don't know what I did wrong.

konopisky, 2021-11-28:
I have the same issue. Or similar issue. I'm able to get to the menu and set time, but after I quit the menu, the time still says 00:00

konopisky, 2021-11-28:
And battery says like 0.3V but my voltmeter says it's 4.2 or something

Metaln00b, 2021-11-28:
Use the branch PCF8263. The newer version of watchy has a different RTC chip.

vimsick, 2021-11-29:
Looks like the pcf8263 branch was missing a change. I have a PR up with the missing deps: https://github.com/sqfmi/Watchy/pull/108

sqfmi, 2021-11-29:
The latest v1.2.8 should resolve the above issues, please create a new issue if issues persists. Thanks!

closed 2021-11-29 by sqfmi