sqfmi/Watchy #118: Long Battery life

DarkZeros, 2021-12-11:
Hi.
I have set myself as goal to lower the battery usage as much as possible (possibly to power watchy from a solar cell).

So far I have submited these 2 PRs.
- Lower 6.5mA/day https://github.com/sqfmi/Watchy/pull/116
- Lower 3mA/day https://github.com/sqfmi/Watchy/pull/117

If you have any other idea, let me know.

I have as further ideas:
- Use voltage regulator circuit to use TPS63900 instead, and have a dropout of 80nA
- Increase the ADC voltage divider resistors

LeonMatthes, 2021-12-16:
One of the best ways that I currently use to reduce battery life is to completely disable my Watchy (put it in deep sleep) between 1am and whenever I wake it up again.

If you wake up at around 8 am every morning, that's 7 hours of almost no battery drain.

That's what it looks like every morning:
[image]

Might be something that you also want to consider to achieve your goal.

DarkZeros, 2021-12-18:
Thanks for the idea!
So far my biggest concern is the base usage (power used while not updatng the display).
With the change to put the ESP32 in lightsleep I already recoverd most of the power usage.

The current state is (as measured):
- Display refresh: Around 40uA (averaged per minute)
- LDO: 40-60uA
- Accelerometer: 100uA
- Voltage divisor: 20uA
- ESP sleep: ? (20uA?)
- RTC sleep: 5uA?
Should be: 130uA + 100uA accelerometer

I my current setup I disabled the accelerometer, and I get around: 160-600uA depending of the display I use.
After a long testing and researching. I found out the reason.
I will open a new issue to notify the devs: https://github.com/sqfmi/watchy-hardware/issues/14

For my watchy I will try to solder a pullup resistor.

LeonMatthes, 2021-12-21:
Wow, an on-time of >1 month was always my hope with watchy. Your efforts are great in that regard.
Do you only have v1 HW available or v2 also?

It would be very interesting to see if you could keep a BLE connection active with v2 in light sleep and see how much power that would draw.

I've tried optimizing my current setup as much as possible, but it still only lasts ~2-3 days with Watchy opening a BLE connection to my phone every minute.
That also unfortunately means I can not use your fix for light sleep during the display update, as that would kill the BLE connection :(

DarkZeros, 2021-12-21:
My approach was to lightsleep and set the wakeup to the BUSY pin. Assuming this was not interfering with anything else.

However, if you want to have a system that works for both continuous BLE and Display, maybe a better idea is to use the FreeRTOS automatic light sleep (RTOS is the underlying mechanism of Arduino tasks, most of the time only 1 task is used)
Here they explain how to set up RTOS to lightsleep automatically, and wake up when BLE needs to update.
https://github.com/espressif/esp-idf/issues/947

Then, you will have to use RTOS HW ISR interrupt for the display, or completely disable the busy callback on the display.
With that approach you can get an average of 2.5mA usage with BLE, That should last around 4 days of battery.

You can also keep the display update code as-is, along as you can detect when BLE is on, and avoid the lightsleep.

LeonMatthes, 2021-12-23:
That would be my ideal setup, unfortunately the Watchy v1 hardware doesn't have the necessary external 32KHz oscillator needed for the BLE + light sleep combination.

Still waiting for someone to confirm they can get automatic light sleep w/ BLE working with such a low power draw on the new v2 hardware.
Would consider ordering a v2 board then...

DarkZeros, 2022-08-03:
Pushed 2 more commits on this same topic:
https://github.com/sqfmi/Watchy/pull/177
https://github.com/sqfmi/Watchy/pull/178

Dop47355-82, 2024-04-08:
Is there a possibility of getting links and instructions on how to get the right solar cells and codes? I'm a complete NOOB, with absolutely zero experience in coding. Would be awesome to build my own energy-harvesting smartwatch.

DarkZeros, 2024-04-09:
Solar cell, any would do as long as the voltage is >4.2V.
The solar cell current output is so small that is not worth dealing with a converter or overcharge protector.
I am connecting the cell directly to the battery, AM-5815 -> https://panasonic.net/electricworks/amorton/lineup/